PiSupply/PiJuice #246: PiJuice dead after updating firmware?

sakura-lain, 2018-12-03:


shawaj, 2018-12-03:
It is unlikely to be totally dead. It seems like probably the firmware flashing failed.

Could you confirm the troubleshooting steps you have already taken?

sakura-lain, 2018-12-05:
Hi!
I mainly followed the steps given [here](https://github.com/PiSupply/PiJuice/issues/105) and [there](https://github.com/PiSupply/PiJuice/issues/108) including changing dtoverlay, on 2 SD-cards with 2 different Raspbian, but with no result at all. As the buttons and leds do not work, I couldn't reproduce the reset steps with SW1 and SW3. Also read [this discussion](https://github.com/PiSupply/PiJuice/issues/175) but it didn't help.

ilyastrodubtsev, 2018-12-10:
@sakura-lain Have you tried to update the firmware not through a gui?

sakura-lain, 2019-03-14:
Hi again and sorry for not answering earlier, but I had much work those last months. I tried again to solve this problem these last days, but even through command line or ssh there's nothing to do, the Rasp cannot see my PiJuice anymore. Any news concerning similar problems ?

Thanks again!

ChristopherRush, 2019-03-14:
I would suggest to follow the guidelines in the FAQ section here - https://github.com/PiSupply/PiJuice/tree/master/Firmware#faq

sakura-lain, 2019-04-22:
Hi !

Yes ! The FAQ solution worked. Thank you !
Unfortunately my previous problem about the Pi refusing to boot or shutting down as soon as it is unplugged has come back. :-(
Anyway this issue is solved, I'll open another one for this shut down problem.

closed 2019-04-22 by sakura-lain