PiSupply/PiJuice #247: PiJuice CLI "non-interactive" to WakeUp PiJuice

DrumSergio, 2018-12-04:
Hi,

I would like to set a WakeUp Alarm every 9:00am Mon-Fri without touching the python CLI GUI. Is that possible? 
I would be configuring PiJuices all automatically via Ansible so the problem here is to make Ansible do what I do by hand without touching this software.

I will set it also to stop every Mon-Fri 5:00pm via crontab, with this script

    #piJuice_stop.py
    from pijuice import PiJuice
    import subprocess

    pj = PiJuice(1, 0x14)
    pj.power.SetPowerOff(20)
    pj.rtcAlarm.SetWakeupEnabled(True)
    subprocess.call(["sudo", "poweroff"])

editing the crontab via a "non-interactive" way using this technique: https://stackoverflow.com/questions/4880290/how-do-i-create-a-crontab-through-a-script#answer-9625233 

Thank you!

tvoverbeek, 2018-12-06:
Yes, Here is python script which sets the alarm as you want:
```
# pjuice_setweekdayalarm.py
from pijuice import PiJuice

pj=PiJuice(1, 0x14)
pj.rtcAlarm.SetAlarm({'second': 0, 'weekday': '2;3;4;5;6', 'hour': 9, 'minute': 0})
```
Use your stop script to shut down the pi to be sure the wakeup-enabled flag is set.

tvoverbeek, 2018-12-06:
Correction: Since the RTC is kept in UTC you need to specify the time in UTC.
So for Spain (UTC+1) the SetAlarm call should be:
```
pj.rtcAlarm.SetAlarm({'second': 0, 'weekday': '2;3;4;5;6', 'hour': 8, 'minute': 0})
```
to get a wakeup at 9 am.
**Note:** You will have to update the script again for Summer time since then Spain is on UTC+2

ChristopherRush, 2019-09-15:
Solution here above - https://github.com/PiSupply/PiJuice/issues/247#issuecomment-444926819

closed 2019-09-15 by ChristopherRush