PiSupply/PiJuice #249: Experimental firmware v1.3

shawaj, 2018-12-08:
Moving this from email into public domain...

After @mmilann coming across a recent paper that states possibility for counting battery charge level based only on battery voltage sampling, he did some experimental software development. 

Here is paper - [IEEKSPC_2017_6_5_20171031173030_1 (1).pdf](https://github.com/PiSupply/PiJuice/files/2659410/IEEKSPC_2017_6_5_20171031173030_1.1.pdf)

PiJuice hardware already has battery voltage signal sampling independently of LC709203F fuel gauge ic and made some algorithm that does estimated current integration what does much more stable charge level without sudden jumps according to experiment. There can be an updated firmware with this new charge measurement as beta version at first so people can test it. Also if this proves, in future fuel gauge ic can be removed lowering cost.

Beta firmware V1.3 is now released for testing... https://github.com/PiSupply/PiJuice/tree/master/Firmware/Beta 

There are two additional scripts:
pijuice-soc-test.py for deriving customised battery parameters for profiles not embedded in firmware.
pijuice-soc-config.py for writing derived profile parameter to PiJuice. 
For existing predefined profiles BP7X 1820, and SNN 2300 parameters are embedded in firmware. 
But for other types it can be derived and written like for a Lifepo4.

For BP7X we had two battery samples and they differs in resistance, one with oval contacts has 
around 230mOhm, other with flat contacts around 155 mOhm. Currently I just just made average of two,
so there can certain inaccuracy because. How ever it can be calibrated writing customised resistance
parameters.

shawaj, 2018-12-08:
Feedback from @tvoverbeek
Ran your test script with my BP7X, 1820mAh using a 10.8 Ohm resistor overnight.
Since I did not have a high wattage resistor I put it together
with eight 22 Ohm 1/4 Watt resistors (4 parallel branches of two 22 Ohm resistors in series).
I have attached my log.
After normalising the dSOC values I got the following:
ocv10	3660
ocv50	3802
ocv90	4064
r10		181.0
r50		183.0
r90		182.0

The battery resistance values were varying but fairly constant over the complete discharge.

Does this look OK or should I have used a smaller test resistor?

Subsequent reply by @mmilann 
```
You results looks as expected. For li-po resistance does not change much with state of charge so it can be even considered 
constant, and can be measured with only few test samples. It will be interesting to see more results from different BP7X samples
and compare results so final parameters can be derived for profile inside firmware.

```

Reply from @tvoverbeek 
```
Some observations on the V1.3 firmware.

After firmware load with pijuiceboot and firmware restart no blue LED 2 as in V1.2

Started charging my 2nd BP7X with the V1.2 firmware. Loaded the V1.3 firmware halfway thru.
Charge level showed started from 0% again, while the Vbat continued increasing.
Only restart with removing battery showed a plausible charge level again.
Switching back to V1.2 showed the good charge level immediately.
Some initiaiization missing in V1.3beta?

```

Final reply from @mmilann 
```
LED indication in V1.2 was for debugging purpose of initial update state and forgotten to remove.

Yes charge level is set to zero after update and it is issue, keeping previous value would be 
better. Fuel gauge ic preserves value inside its memory  in V1.2 and hence no reset to zero.
```

shawaj, 2018-12-11:
More updates....

The false indication of zero charge level after firmware restart should be fixed.
Milan, any way of fixing (and testing the fix) for this?
I do not know if more data for the tables is needed in 1.3, especially
for the supported non BP7X batteries.
Otherwise the big discrepancies between charging and discharging behavior are gone.

And reply...

There are recent firmware update regarding new pijuice data hat, with possibility to receive wake-up interrupt via IO2 on P3 from modem. Also update 
with possibility to measure  battery temperature via NTC and state of charge without soldered fuel gauge ic and bad charge init after update is fixed. 
It will automatically detect if not soldered or is in bug state and than switch estimation on mcu. New firmware can be released soon if needed to integrate 
with new package version, also some GUI/API updates need to do required for updated firmware easy configuration.

shawaj, 2018-12-15:
Also @mmilann has created many commits on the develop branch:

https://github.com/PiSupply/PiJuice/commit/e758f1c757fa4ef81410fac88b08bb4605cbde25
https://github.com/PiSupply/PiJuice/commit/aee8b506aeb82cf7a5d342d920331986a256ec55
https://github.com/PiSupply/PiJuice/commit/bb253c0f4ab7b75b5c39fddb2663df7dcbd8736c
https://github.com/PiSupply/PiJuice/commit/aa92c0b1c966a00f9d3ec8bd2db86febbc7236be
https://github.com/PiSupply/PiJuice/commit/15db731521433cfb2032810e8477b37a762e178b
https://github.com/PiSupply/PiJuice/commit/8f442344315ad1056fef3d56292614ec16e9fdfd

@wikipedia555 @ilyastrodubtsev can you sort the diffs to bring the changes into the main branch?

jdotbdot, 2018-12-15:
> The false indication of zero charge level after firmware restart should be fixed.
Not for me:
I downloaded and installed firmware 1.3 this morning. The indicated battery level went from 100 to 0.
Rebooting the Pi made no difference.
Rebooting once without battery and again with, it showed 80, rising slowly to 98 (so far)

shawaj, 2018-12-15:
@jdotbdot have you used the latest firmware in my above comment?

jdotbdot, 2018-12-15:
No, PiJuice-V1.3_beta_2018_09_13.elf.binary. Sorry, my mistake!

shawaj, 2018-12-15:
I need to update that one in the main repo but wasn't by a computer yesterday. I'll get everything updated over the weekend.

Let us know how you get on with the new one though!

jdotbdot, 2018-12-16:
Two graphs of reported battery charge vs time with firmware 1.2 and 1.3 on a Raspberry Pi 3A+.
The Pi uses Node-Red and scripts to acquire the battery status every 10 seconds and send it by MQTT over wifi. Otherwise it is idle.

[image]

[image]

The graph for firmware 1.3 is clearly much more linear. It does not show an immediate drop when the USB power is removed, nor the strange and presumably spurious downward spikes. 

The Pi stayed alive for about half an hour longer with v1.3. Maybe the overreported voltage when charging with v1.2 stops charging sooner than necessary?

If the PiJuice had been set to shutdown at battery level 3, the dips shown with firmware v1.2 would have shut it down after just 2 hours. With 1.3 it reached battery level 3 after 5 hours.
The Pi is living on the edge at these low levels though, if you run extra code and exercise the processor, the battery is likely to die and the Pi goes down without a proper shutdown.

The graph when charging is similarly linear with 1.3 too, though I didn't take a snapshot of it.

Conclusion, based on just one comparison, is that firmware 1.3 is much better. No doubt the profile could be adjusted to charge it more on 1.2? 
Does the battery contain circuitry to protect it from over charging as well as over discharge?

ilyastrodubtsev, 2018-12-17:
@shawaj Yes OK. I can merge two branches.

shawaj, 2018-12-17:
@ilyastrodubtsev ok thanks. @tvoverbeek has created new v1.4  deb files from the latest codebase here - https://github.com/PiSupply/PiJuice/tree/master/Software/Install

I am signing these and uploading to Raspbian repo as we speak. So they should be available soon. This still has the v1.2 firmware.

We will then upgrade the software to work with new changes for v1.3 firmware and update software version to 1.4.1 after that.

Let us know when you have merged them and ready for testing.

shawaj, 2018-12-19:
@tvoverbeek @ilyastrodubtsev @ChristopherRush i have now, i think, updated the develop branch to where it should be now. 

could you take a look?

tvoverbeek, 2018-12-20:
When updating the firmware from 1.2 to V1.3_2018_12_14 I noticed that after the update the Pi is switched off (this is with power connected to the PiJuice microUSB connector).
This is a change from before. When loading the 1.2 firmware itjsut starts after the load.
This breaks the 'Wait for firmware restart' after the firmware upload in the gui/cli.
@mmilann Is this intentional? If so we have to change the gui and cli. I would prefer the firmware jsut (re)starts after upload.

shawaj, 2019-01-15:
Here is the changelog - [Change log.txt](https://github.com/PiSupply/PiJuice/files/2760495/Change.log.txt)

tvoverbeek, 2019-01-15:
Thanks. Will test the 2019-01-15 beta firmware

shawaj, 2019-01-15:
Thanks @tvoverbeek

GizmoEV, 2019-02-11:
Is there much risk of a permanent bricking of the PiJuice by installing this beta firmware? I don't mind a worst case of having to follow the procedures outlined in the FAQ for unbricking a successfully installed firmware.

tvoverbeek, 2019-02-11:
@GizmoEV No, I have been going back and forward between the 1.2 and the beta 1.3 from 2019-01-15 a few times without any problems.

GizmoEV, 2019-02-12:
@tvoverbeek Thank you. I'll give it a try.

shawaj, 2019-02-28:
Updated firmware with new battery profiles added - [drive-download-20190228T171448Z-001.zip](https://github.com/PiSupply/PiJuice/files/2916077/drive-download-20190228T171448Z-001.zip)

shawaj, 2019-02-28:
New message from @mmilann 

```
Hi All,

test on second set of batteries shows pretty similar battery characteristic and now parameters are extracted and embedded in firmware source. Now released updated V1.3_2019_02_28 firmware with addition of new predefined battery profiles.
https://drive.google.com/drive/u/1/folders/1Io6A7HZZLLdI4zwE-PYkMlbMaZCpYBhh

Also it requires software update of battery list in pijuice.py api to:
batteryProfiles = ['BP6X_1400', 'BP7X_1820', 'SNN5843_2300', 'PJLIPO_12000', 'PJLIPO_5000', 'PJBP7X_1600', 'PJSNN5843_1300', 'PJZERO_1200', 'PJZERO_1000', 'PJLIPO_600', 'PJLIPO_500']
```

ChristopherRush, 2019-09-15:
Not experimental anymore, closing.

closed 2019-09-15 by ChristopherRush