PiSupply/PiJuice #250: needExternalOnOffButton+++

lidense, 2018-12-11:
Hi,
I find the SW1-SW3 extremely difficult to use because too small, especially if you want to house the pijuice in a box.
I looked for information on how to add an external push button for example for on/off but found nothing.
Please advise
Thanks 
Dave

shawaj, 2018-12-11:
This is exactly what J5 header is for.

See here https://github.com/PiSupply/PiJuice/blob/master/Hardware/README.md#buttons

You just need to solder wires to that header and it works the same as SW1. Or you could solder a header and then have the button removable

closed 2018-12-11 by shawaj

lidense, 2018-12-11:
OK thanks, that's great news as SW1 is the most important button ! Anything similar for SW2 and SW3?

shawaj, 2018-12-11:
@lidense no there is nothing like that for SW2 and SW3 - you would have to remove the buttons and solder wires striaght to the pads on the PCB