PiSupply/PiJuice #251: PiJuice on Windows IoT Core

tresoft, 2018-12-14:
Hi PiJuice team.

I use Windows IoT core on the Raspberry PI 3 and would like to use PiJuice, because here the power supply via GPIO is possible. So far so good.

How is it possible without the Python libraries to read the following values:

-Charge of the battery
-Mains power or emergency power
-Error
-Date and time of the RTC

It is actually to read out via I2C? Could you provide a few details to do this? Thank you very much.

Greetings from Germany
Stephan

shawaj, 2018-12-19:
That's really cool. Mind if we link to it etc from our guides?

CC @ChristopherRush

tresoft, 2018-12-19:
Yes, please!

shawaj, 2018-12-19:
Added it to our readme - https://github.com/PiSupply/PiJuice/commit/1cd62fc19e0966faccdafc016d5476dba64fa042