PiSupply/PiJuice #256: PiJuice sometimes won't power on after SYS_FUNC_HALT_POW_OFF

TylerAdkisson, 2018-12-26:
Hi,

We have 18 PiJuices in use, and we've noticed that here and there that some won't power back on after a graceful power off (via `SYS_FUNC_HALT_POW_OFF`).  When in this state, pressing SW1 (`SINGLE_PRESS` set for `HARD_FUNC_POWER_ON`) causes the PiJuice's green LED to turn on for about a second or two, then back off; never powering the pi.  To get it to power back on properly we have to hold SW1 for 10 seconds (`LONG_PRESS1` set for `HARD_FUNC_POWER_OFF`), then we can power it on using SW1 normally.  The behavior appears random, and I've been unable to consistently reproduce the issue.
This seems to be the same problem reported in #196, which was never resolved.

Additional information:
Battery charge nor external power appear to have an effect: It's happened with a fully charged and partially charge battery, USB power plugged in or not.
PiJuices with firmware versions 1.1 and 1.2 have exhibited this issue.

To me, it seems something like the PiJuice still thinks it's in the powered on state after cutting the 5v rail, and thus ignores the `HARD_FUNC_POWER_ON` request, instead showing battery level via the LED color like when powered on.

fencer, 2019-04-28:
I have the exact same issue on our prototype! This has manifested its self in the majority of our tests up to the extent that we are now considering an alternative unless addressed ASAP. We are using firmware 1.2 but was also recorded in version 1.1

tvoverbeek, 2019-04-28:
Could you try with the latest 1.3 Beta firmware from the develop branch:
https://github.com/PiSupply/PiJuice/blob/develop/Firmware/Beta/PiJuice-V1.3_2019_02_28.elf.binary?raw=true
To see if the problem is still there.

shawaj, 2019-04-29:
From @mmilann :

mcu checks communication over I2C timeout of 15 seconds before turn-on, also it checks if 5V regulator is on or power on rpi present, on older firmware it checks also charge level.

Also latest firmware is recommended. 

It can be considered to open new firmware development branch (2.0) with freertos introduction. Os can make code more clean and easier for others to review things like power management functions, and maybe to customise it, if becomes stable it can be official.

ChristopherRush, 2019-09-15:
Suggestions above, open if not satisfied.

closed 2019-09-15 by ChristopherRush