PiSupply/PiJuice #257: Raspberry Pi A+, to Pogo Pin or not to Pogo Pin?

robcee, 2018-12-29:
Since the PI A+ is relatively new, I think the PiJuice documentation is out of date.

Looking at the A+ board, it looks like there is a RUN pad similar to the 3B that should line up with TP1 on the PiJuice. I haven't tested it yet because I wanted to verify this was the correct placement before going ahead with it.

shawaj, 2018-12-29:
@ChristopherRush @tvoverbeek @mmilann have you tried this?

From pictures it looks like it's in a different place to all existing pi models but hard to tell without them in front of me

robcee, 2018-12-30:
I will take some pictures and report back!

shawaj, 2018-12-30:
I'll be in the office tomorrow too so will have a play

shawaj, 2019-01-11:
@robcee the cable option would also work but makes it harder to take off in a hurry

robcee, 2019-01-15:
@shawaj thanks for the update. I'll stay tuned for progress on the pin shorting solution as that sounds like a good solution.
I was considering adding a connector to any leads I soldered on, but obviously that's still not ideal.

shawaj, 2019-01-15:
@robcee see the updated firmware with this functionality attached here 
[V1.3_2019_01_15-20190115T155652Z-001.zip](https://github.com/PiSupply/PiJuice/files/2760365/V1.3_2019_01_15-20190115T155652Z-001.zip)

Could you try it out?

robcee, 2019-01-15:
> Could you try it out?

I sure can! I’ll report back.

shawaj, 2019-01-15:
@robcee great! On the GUI, just leave the option for the pogo pin as uninstalled, and it will default to this new behaviour.

Now we can support both.

tvoverbeek, 2019-01-16:
Running a first test with the V1.3_2019_01_15 firmware. Power connected to RPi USB.
Run pin at not installed.
After shutting down the RPi a single SW1 press wakes it up again.
So the SCL wakeup works.

robcee, 2019-01-17:
Also confirmed. I've been using this for a couple of days now and everything seems stable.

Long press sends a SHUTDOWN_HALT to the pi and it shutsdown gracefully (still receiving power). Single press boots it back up.

Thanks for this!

shawaj, 2019-01-17:
Great, thanks @robcee thanks for the update. Will close this now but feel free to reply back if you notice anything weird

shawaj, 2019-01-17:
@ChristopherRush could you document this at some point?

closed 2019-01-17 by shawaj

robcee, 2019-01-17:
@shawaj thanks Aaron. that works for me.

One question, does this mean pin 5 is unavailable with this in place?

shawaj, 2019-01-17:
@robcee nope. Because we use pin 5 for i2c for the PiJuice to communicate with the Pi. When the PiJuice / Pi is on they are used for that

ChristopherRush, 2019-01-18:
@shawaj @tvoverbeek Does this method also work for WakeupAlarm?

shawaj, 2019-01-18:
It should do yes. I haven't tested it though

tvoverbeek, 2019-01-18:
@ChristopherRush @shawaj 
Just tested it. Power connected to RPi. It works.
Note you need the 1.3 beta firmware from Jan 15, 2019:
https://github.com/PiSupply/PiJuice/blob/develop/Firmware/Beta/PiJuice-V1.3_2019_01_15.elf.binary