PiSupply/PiJuice #258: running pijuice_cli as root gives i2c error

TobbeJ, 2018-12-31:
Hi

If i try to start pijuice_cli as root then i get an error about not being able to find the pijuice over i2c but if i run it as regular user it works.

Why should i not be able to run the cli as root?

tvoverbeek, 2018-12-31:
I could reproduce your error. It is caused by a permission error when trying to open i2c bus 1 in the smbus module. This module is a C-extension (shared library), not a python script.
pijuice_cli is running as user pijuice (non privileged user) since version 1.4, even when you start it with sudo.
However, after some testing the error disappeared for me. I could then run pijuice_cli from the pi account and with sudo without problems.
Do you have other programs running using the i2c bus (=loading the smbus module)?

TobbeJ, 2019-01-02:
no other program running using i2c that i'm aware of.
this was with a fresh install of rasbian and only thing i've done is to install pijuice (not the gui, just cli since it is a headless pi)

then as root i tried to run the cli and it failed.

tvoverbeek, 2019-01-03:
@TobbeJ Found a solution.
First, it should not be necessary to run pijuice_cli as root.
You can do everything as a normal user.
If you insist to run as root, you need to add the root user to the i2c group: `sudo addgroup root i2c`.
You can check the i2c entry in /etc/group if root is part of the i2c group.
With root as member of the i2c group `sudo pijuice_cli` works.

tvoverbeek, 2019-01-17:
Closing issue.
@TobbeJ If you do not agree, reopen it.

closed 2019-01-17 by tvoverbeek

webghostx, 2019-02-11:
same issue with Dietpi on Raspi 3 B+ .
`addgroup root i2c` did not work.

I2C looks good, right?
```
root@IoT-Server:~# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --    
```

teeso, 2019-08-30:
Hi, maybe someone can help me I still got message that No PiJuice found or failed to use i2c, I done all of the above but no way to works 

i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- 5c -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 

thanks in advance
Carlo

tvoverbeek, 2019-08-30:
@teeso Why are you polling i2c bus 3 (a bit-banged i2c bus)
What do you have at address 0x5C? This is not an address used by PiJuice.
Which GPIOs/pins are you using for bus 3?
PiJuice uses the 'normal' i2c bus 1 on GPIO 2 and 3, pins 3 and 5.
What is the output of `i2cdetect -y 1`?