PiSupply/PiJuice #260: RTC wakeup

pabou38, 2019-01-07:
Hello ..
I am a customer of pijuice. from the original kickstarter. I'm trying to wake up the PI every 20mn, do some processing, then cut power and halt.

My timezone is paris, so the RTC is set (thru GUI) to one hour less than system time (the RTC is displayed in the GUI).

I'm setting the RTC alarm thru the API. minutes is set to current minute (from datetime) plus 20 modulo 60. seconde is set to zero. hours and days to every.
I'm also enabling wakeup thru API.

I observe a wakeup every 1h20, instead of every 20mn. I suspect a problem around UTC, but I have no clue

I'm also cutting power with setpoweroff , with a 20sec delay to let the halt command execute. Another issue is that from time to time, after halt and "poweroff", the PI red led is still on (as well as the green one). So not sure if power is actually off, but I see a quick battery drain,  so I suspect it is not.  and my apps is solar powered, so I need to save energy.

I'm on firmware rev 1.2. PI2.

tvoverbeek, 2019-01-07:
@pabou38 Pascal, when you specify the alarm with the API it has to be specified using UTC.
So if you used system time (CET, Paris time) to set the next alarm it would explain the 1h20m you see.

Regarding the setpoweroff, is this with the new function under system tasks (Software Halt Power Off) in the 1.4 release (since SYS_FUNC_HALT_POW_OFF uses a 60 second delay before removing the 5V power). With both Pi's red and green leds on it is certainly not off but still active. Normally green led on means read/write activity to the SD card is going on.

tvoverbeek, 2019-01-07:
@pabou38 1.4 is the raspbian package version.
The firmware is still at 1.2
When you use the API for poweroff, do not forget to set the wakeup enable before you send the 'sudo halt'.

closed 2019-09-16 by ChristopherRush