PiSupply/PiJuice #262: Voltage feedback on J4 Terminals

drv2e30, 2019-01-08:
So I attempted to use the J4 terminal to power the unit via a spliced USB cable. It appeared to work fine for a short while before the Windows machine or a USB bus states that a USB power surge is occurring on the bus and it then shuts the ports and the bus off. Has anyone experienced this?

shawaj, 2019-01-08:
@drv2e30 are you saying that you are trying to do this from a USB port on another machine? i.e. you are using the USB port on a windows PC to try and charge pijuice?

drv2e30, 2019-01-09:
Yes, I'm using a USB cable with a USB A connector attached to a windows PC with the other end of the cable cut off and a positive and negative connection attached to the pins on J4.

shawaj, 2019-01-09:
@drv2e30 and you are getting the error on the windows PC about the power surge?

drv2e30, 2019-01-10:
I get this error on the Windows pc. Ive also tried to use a usb bus that would shut the port off once I connected the pi juice.

ChristopherRush, 2019-09-15:
I'm guessing the solution here is not to use power from a PC/Laptop. I'm would assume there might be a current limit too 500mA?. Open if you disagree.

closed 2019-09-15 by ChristopherRush