PiSupply/PiJuice #263: Supply voltage detection level

tresoft, 2019-01-08:
Hello,

currently I have 6 RPI3 running with PiJuice. Of these, two run with Raspian and the rest with Windows IoT Core. 

For half of the devices, the GPIO voltage is under full load at about 4.95 volts and all is well.

For the other half, the voltage is about 4.85 volts and the batteries are not charged. If the voltage returns after a voltage failure, then it will no longer be switched to main power mode. 

This behavior is independent of the operating system.  Its occurs on Linux as well as Windows (See photos).

Since I cannot change the used power supplies: Is there a possibility of adjusting the threshold at which the supply voltage is detected?

Greetings
Stephan

[image]

[image]

tresoft, 2019-01-08:
I don't expect the metric readings to match PiJuice. The slight deviation is perfectly fine. The photos are only meant to show that PiJuice measures accurately. 
The problem is the rabor dance of the starting voltage in the power parts, which is also completely normal. The power supplies are fixed and considered as a black box.

It can always happen during the lifetime of such an application that tensions change within certain limits. PiJuice would have to be more generous at this point.

All devices with a GPIO voltage smaller than 4.9V do not charge the battery. With a voltage greater than 4.9V, everything is ok. Can this threshold be parametrized to 4.8V, for example?

ChristopherRush, 2019-09-15:
In this scenario the power is provided by the Raspberry PI's micro USB to the PiJuice via 5V GPIO and as such is rated at a minimum of 4.8V. If power is supplied to PiJuice micro USB it has a much wider voltage range, unless i'm missing the point here.

closed 2019-09-15 by ChristopherRush