PiSupply/PiJuice #264: No snap availabel

webghostx, 2019-01-08:
has anyone installed pijuice-base on ubuntu core snappy?
I have had no success with apt or dpkg so far

ChristopherRush, 2019-01-09:
The PiJuice packages are only available on the Raspbian APT. You would have to manually install it using this GitHub repo - https://github.com/PiSupply/PiJuice/tree/master/Software/Install

shawaj, 2019-01-09:
@webghostx if you do do this then please let us know and we can link to it from the front page of the repo

shawaj, 2020-12-05:
@webghostx did you manage to do this?

DannyQuantum, 2021-02-19:
Just dropping a comment so I can find my way back here in a couple of days - I'm porting my projects to Ubuntu Core over the next couple of days, including the PiJuice so will be working on this myself. Will keep you posted!