PiSupply/PiJuice #265: pijuice-base : Depends: python3 (>= 3.5) but 3.4.2-2 is to be installed

rburgul, 2019-01-09:
I get the following error when trying to install on jessie. I have python 3.7 installed.

`$ sudo apt-get install pijuice-base
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 pijuice-base : Depends: python3 (>= 3.5) but 3.4.2-2 is to be installed
E: Unable to correct problems, you have held broken packages.`

ChristopherRush, 2019-01-09:
What does `python -V` show?

tvoverbeek, 2019-01-09:
@rburgul I suppose you installed python3.7 with PIP.
Then the Raspbian package database is not aware of it.

Can you upgrade to stretch?
To upgrade to stretch:
Change /etc/apt/sources.list to:
```
deb http://raspbian.raspberrypi.org/raspbian/ stretch main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ stretch main contrib non-free rpi
```
and /etc/apt/sources.list.d/raspi.list to:
```
deb http://archive.raspberrypi.org/debian/ stretch main ui
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ stretch main ui
```
and run `sudo apt-get update` followed by `sudo apt-get dist-upgrade`
Works best for headless (raspbian-lite).
For the desktop it is probably better to start from scratch.

rburgul, 2019-01-09:
Thanks both.
python -V says Python 2.7.9
python3 -V says 3.7.2
I installed python 3 by downloading and make, sudo make altinstall.
This seemed to work.
It is a headless setup. I'll try the stretch update..

rburgul, 2019-01-10:
in the end I just did a clean install of stretch and got up and running.

shawaj, 2019-01-10:
Awesome glad to hear it!

closed 2019-01-10 by shawaj

StudioBeneden, 2019-02-25:
I'm having the same issue, Im on jessie and can't (easily) upgrade to stretch because of another install that's dependant on Jessie.. I've tried downgrading Python to 3.4.2 manually (python3 -V shows: Python 3.4.2), but the piJuice installer still refuses to run with the same warning. Hope there's a way around this?