PiSupply/PiJuice #268: COMMUNICATION_ERROR in PiJuice Settings

slead, 2019-01-12:
I received one of the original KickStarter PiJuice units and only just got around to testing it now.

I have a Raspberry Pi B2 running Stretch and updated using `sudo apt-get update` and `sudo apt-get dist-upgrade` then installed the PiJuice software using `sudo apt-get install pijuice-gui`.

I can see the PiJuice Settings app, but it's showing COMMUNICATION_ERROR, similar to https://github.com/PiSupply/PiJuice/issues/175

[image]

The PiJuice *does* correctly supply power to the Raspberry Pi if I unplug both from the micro USB.

`i2cdetect -y 1` shows detected connections for the addresses at 14 and 68:

[image]

I haven't made any changes to the PiJuice (firmware, etc) since getting it in the post. I've ascertained that it is correctly seated on the 40-pin connector.

Thanks for any advice.

tvoverbeek, 2019-01-12:
@slead Could you try the following:
After a boot with the PiJuice installed run `i2cdetect -y 1` again.
If you then see 'UU' instead of '68' at position 0x68 then at least the HAT
eeprom is read correctly (provided its address was not change).
'UU' means the Raspbian real-time-clock module is loaded as specified
in the device-tree overlay loaded from the HAT eeprom.
Let me know the result and we can take it from there.

slead, 2019-01-12:
After rebooting the Raspberry Pi with the PiJuice software installed and the HAT in place (fully charged with a green LED) there is no change in `i2cdetect -y 1` which still reads 68.

The PiJuice is still correctly supplying power to the Pi when there's no micro USB, but the PiJuice Settings app still shows COMMUNICATION_ERROR.

Thanks

tvoverbeek, 2019-01-13:
Something is strange. The window header in your screen shot says 'Pijuice Settings'.
Since the latest Raspbian package version (1.4) pijuice_gui only runs with python3.
The window header should say 'Pijuice Settings (Python 3 version)'
[image]
Are you sure you are running stretch and have the 1.4 package version installed?
What is the output from `lsb_release -a`?
My output is:
```
pi@rpi-pijuice:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.6 (stretch)
Release:	9.6
Codename:	stretch
pi@rpi-pijuice:~ $ 
```
to check the installed package version use `apt list --installed 'pijuice-*'`
My output:
```
pi@rpi-pijuice:~ $ apt list  'pijuice*'
Listing... Done
pijuice-base/stable,now 1.4 all [installed,automatic]
pijuice-gui/stable,now 1.4 all [installed]
pi@rpi-pijuice:~ $ 
```

slead, 2019-01-14:
Thanks for the fast responses.

The screenshot in my first post was taken before I reinstalled Stretch and reinstalled PiJuice - after I reinstalled both, my PiJuice Settings app now lists 'Python 3 version' as in your screenshot.

The output of `lsb_release -a` and `apt list --installed 'pijuice-*'` is identical to your versions:

[image]
[image]

But the problem with COMMUNICATION_ERROR remains unchanged in the PiJuice Settings app. Thanks

tvoverbeek, 2019-01-14:
Since you get an i2c device detected at address 0x14 my suggestion is to try to flash the current firmware with the pijuiceboot program.
You will find the firmware in /usr/share/pijuice/data/firmware.
To be safe stop the pijuice service (suspect it failed anyway because of the commnication errors):
`sudo systemctl stop pijuice.service`
No pijuice_gui and no pijuice_tray should be running (I also suspect the tray app failed because of the comm errors). Kill them if necessary.
Then in a terminal
`pijuiceboot 14 /usr/share/pijuice/data/firmware/PiJuice-V1.2_2018_05_02.elf.binary`
Output should look like this:
```
pi@rpi-pijuice:~ $ pijuiceboot 14 /usr/share/pijuice/data/firmware/PiJuice-V1.2_2018_05_02.elf.binary
using address 14
Input file /usr/share/pijuice/data/firmware/PiJuice-V1.2_2018_05_02.elf.binary
Starting bootloader 
Starting bootloader old firmware
Sending get command 
bytesRead 13  dataLen 11
ack 79
data: b, 10, 92
data received 11
bootloader version: b
commands: 10, 0, 1, 2, 11, 21, 31, 44, 63, 73
erase page count 35
first page erase succcess 0
erase success 0
page count 276
mem data read 256, 256
Page 275 programmed successfully
mem data read 256, 256
Page 274 programmed successfully
mem data read 256, 256
Page 273 programmed successfully
mem data read 256, 256
Page 272 programmed successfully
....
....
Page 5 programmed successfully
mem data read 256, 256
Page 4 programmed successfully
mem data read 256, 256
Page 3 programmed successfully
mem data read 256, 256
Page 2 programmed successfully
mem data read 256, 256
Page 1 programmed successfully
mem data read 256, 256
Page 0 programmed successfully
EEPROM programming finished successfully
Code executed successfully
pi@rpi-pijuice:~ $ 
```
The firmware (if load was successful) will restart and with this firmware version LED 2 will light bright blue.

Hope this works, if not I guess it is time to contact Pi-Supply for a replacement with reference to this issue.

slead, 2019-01-15:
I tried stopping the service and updating the firmware but I got `error receiving data -1`. I ran a ping to confirm that I have internet.

I guess this looks like a hardware fault then? Could it be due to my being based in Australia - could it be trying a non-existent mirror or something?

[image]

I'll try PiSupply to see whether they can replace it as faulty if you agree that it's a hardware fault. Thanks very much for the help in debugging this.

shawaj, 2019-01-15:
Thanks for the update @slead