PiSupply/PiJuice #269: Crash when setting custom battery profile

rburgul, 2019-01-12:
New raspian stretch-Lite system on headless raspberry pi 2. Got 12000mAh battery wired in and pijuice software installed and running.
Only issue is when I try I set the battery profile to custom and change any parameter eg mAh, I get the following. Also included i2cdetect result. Any help greatly appreciated:
`$ pijuice_cli.py
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 1320, in <module>
    loop.run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 719, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python2.7/dist-packages/urwid/raw_display.py", line 393, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python2.7/dist-packages/urwid/raw_display.py", line 493, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 403, in _update
    self.process_input(keys)
  File "/usr/lib/python2.7/dist-packages/urwid/main_loop.py", line 503, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python2.7/dist-packages/urwid/container.py", line 592, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python2.7/dist-packages/urwid/container.py", line 1587, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python2.7/dist-packages/urwid/container.py", line 2269, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python2.7/dist-packages/urwid/decoration.py", line 621, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python2.7/dist-packages/urwid/listbox.py", line 987, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python2.7/dist-packages/urwid/decoration.py", line 621, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python2.7/dist-packages/urwid/wimp.py", line 535, in keypress
    self._emit('click')
  File "/usr/lib/python2.7/dist-packages/urwid/widget.py", line 464, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python2.7/dist-packages/urwid/signals.py", line 264, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python2.7/dist-packages/urwid/signals.py", line 294, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 970, in _apply_settings
    status = self.write_custom_values()
  File "/usr/bin/pijuice_cli.py", line 992, in write_custom_values
    profile['capacity'] = int(self.param_edits[0].value())
AttributeError: 'Padding' object has no attribute 'value'
$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- 44 -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --         
`

shawaj, 2019-01-12:
Do you know what version of the software and firmware you are using?

shawaj, 2019-01-13:
I assume you mean the firmware is 1.2? Can you check the software version too?

rburgul, 2019-01-13:
it was 1.3.3. Just upgraded to 1.4 and rebooted but still get this behaviour:

`$ pijuice_cli
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 1808, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 719, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 393, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 493, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 403, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 503, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 592, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1587, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2269, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 621, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 985, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 621, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 535, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 463, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 264, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 294, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1087, in _apply_settings
    status = self.write_custom_values()
  File "/usr/bin/pijuice_cli.py", line 1109, in write_custom_values
    profile['capacity'] = int(self.param_edits[0].value())
AttributeError: 'Padding' object has no attribute 'value'
`

shawaj, 2019-01-13:
Can you confirm the exact steps to reproduce this error message?

tvoverbeek, 2019-01-13:
@rburgul This is a bug in piuice_cli.py not discovered before because nobody (?) defined a custom battery profile with pijuice_cli. WIll try to fix it. Will also check if the same bug exists in pijuice_gui.
Need to use it myself too since I also have the 12000mAh battery.
The code area (last items in the trace back) in pijuice_cli.py was not updated between 1.3.3 and 1.4, hence you got the same crash.

tvoverbeek, 2019-01-13:
@rburgul Fixed the issue in pijuice_cli.
Replace your version of /usr/bin/pijuice_cli.py with the one from the develop branch:
https://raw.githubusercontent.com/PiSupply/PiJuice/develop/Software/Source/src/pijuice_cli.py
Make sure the owner,group of pijuice_cli.py is pijuice,pijuice
If not `sudo chown pijuice:pijuice /usr/bin/pijuice_cli.py`

rburgul, 2019-01-13:
Brilliant that worked. No crash.
2 things:
1. The fields requested in the software are different to the values given on the website. The website does not specifically state values for  regulation voltage, Cutoff voltage, cold, cool, warm, hot temps, NTC etc
2. I change Charge current to 2400 but it changes to 2350 when I check. Likewise termination current changes itself from 600 to 400.
Thanks for fixing so quickly.

tvoverbeek, 2019-01-13:
@rburgul For some background read the Battery Profiles section in the github Hardware readme.
Regulation voltage (4.18V) and Cutoff voltage (3.0V) are usually the same for all single cell LiPo batteries.
Regarding the 2400 vs 2350. PiJuice stores the charge current with a reduced number of bits
The stored value is (input - 550) // 75 (Integer division).
When read back the value presented is val * 75 + 550.
If you work thisfor 2400 you see why you get 2350.
Look at the code in pijuice.py for storing/reading the battery profile

rburgul, 2019-01-13:
That makes sense.
Thank you so much for your time and patience... on a sunday!

closed 2019-01-13 by rburgul

tonynoel, 2019-02-08:
Hi @tvoverbeek your fixed dont work for me. 
> @rburgul Fixed the issue in pijuice_cli.
there is this return:

tvoverbeek, 2019-02-08:
@tonynoel Need more info about how you got the error.
Is there a simple way to provoke the error?
Then I can trace where it comes from.

ChristopherRush, 2019-02-11:
@tvoverbeek When I set charge current to 240mA it reverts to 550mA for a 1200mAh battery.

tvoverbeek, 2019-02-12:
@ChristopherRush You wrote
> When I set charge current to 240mA it reverts to 550mA for a 1200mAh battery

I do not have a Pijuice Zero, but I assume it uses the same firmware as the PiJuice HAT.
For an explanation of what you see see my comment from Jan 13 above:
> PiJuice stores the charge current with a reduced number of bits
The stored value is (input - 550) // 75 (Integer division).
When read back the value presented is val * 75 + 550.

If you want a lower threshold @mmilann has to modify the firmware