PiSupply/PiJuice #270: i was having problems with the reboot

klint-k, 2019-01-17:
Hello

I was having problems with reboot so here is a script that is called hardreboot.

`#!/usr/bin/python3
# -*- coding: utf-8 -*-
from __future__ import print_function

import time
import pijuice
import subprocess
import datetime
import os
import sys 

DELTA_MIN=1

#sudo apt-get install python-smbus
#sudo apt-get install i2c-tools

# Rely on RTC to keep the time
#subprocess.call(["sudo", "hwclock", "--hctosys"])
##
### Record start time
##txt = datetime.datetime.now().strftime('%Y-%m-%d %H:%M:%S') + ' -- Started\n'
##with open('/home/pi/test.log','a') as f:
##    f.write(txt)

# This script is started at reboot by cron.
# Since the start is very early in the boot sequence we wait for the i2c-1 device
##while not os.path.exists('/dev/i2c-1'):
##    time.sleep(0.1)

def time_a():
    # Set RTC alarm 5 minutes from now
    # RTC is kept in UTC
    t = datetime.datetime.utcnow()
    a={}
    a['year'] = t.year
    a['month'] = t.month
    a['day'] = t.day
    a['hour'] = t.hour
    a['minute'] = (t.minute + DELTA_MIN) % 60
    a['second'] = t.second
    return a

def delay():
    # Do the work
    for i in range(60):
       print('*', end='')
       sys.stdout.flush()
       time.sleep(1)
    print()

def set_power():
    status = pj.rtcAlarm.SetAlarm(time_a())
    if status['error'] != 'NO_ERROR':
        print('Cannot set alarm\n')
        sys.exit()

def Help(h_print = ''):
    if h_print != '':
        print()
        print(h_print)
        print()
    print('This is help')
    print()
    print('Usage:')
    print('  sudo hardreboot [options]')
    print('')
    print('General Options:')
    print('  -h, --help                  Show help.')
    print('  -d  --delay                 Delays the restart with a display')
    print('')

if __name__ == '__main__':
    if getuid() != 0:
        exit(Help('Must have super user powers!'))
    try:
        pj = pijuice.PiJuice(1, 0x14)
    except:
        print("Cannot create pijuice object")
        sys.exit()
    held_for=0.0
    Argv = []
    for A in range(1, len(argv)):
        U = argv[A]
        Argv.append(U)
    if len(Argv) == 0:
        set_power()
        # Enable wakeup, otherwise power to the RPi will not be
        # applied when the RTC alarm goes off
        pj.rtcAlarm.SetWakeupEnabled(True)
        time.sleep(0.4)
        # PiJuice shuts down power to Rpi after 20 sec from now
        # This leaves sufficient time to execute the shutdown sequence
        pj.power.SetPowerOff(20)
        subprocess.call(["sudo", "poweroff"])
    elif '-h' in Argv:
        Help()
    elif '--help' in Argv:
        Help()
    elif 'help' in Argv:
        Help()
    else:
        for arg in range(len(Argv)):
            if Argv[arg].startswith('-'):
                if Argv[arg] == "-d" or Argv[arg] == "--delay":
                    set_power()
                    pj.rtcAlarm.SetWakeupEnabled(True)
                    time.sleep(0.4)
                    pj.power.SetPowerOff(20)
                    subprocess.call(["sudo", "poweroff"])`

shawaj, 2019-01-17:
What problems were you having?

klint-k, 2019-01-17:
It seemd that there was a problem with the restart. the 7" screen would freeze about 90% to working and I could not use it.

klint-k, 2019-01-17:
ps. Is there a PiSupply cross platform Issues?

shawaj, 2019-01-17:
No there isn't. Only the individual repo ones currently

tvoverbeek, 2019-01-17:
@klint-k Some comments on your script.
You have a problem when the UTC time is xx:59.
Adding 1 minute will put the time in the past.
Better to use Python a timedelta object of DELTA_MIN minutes and add it to the current utc time before generating a{}.
Also as a safety you might want to set the RTC to utcnow(). Now you are relying on the RTC being in sync with the system clock.
Your delay() function is not called when the delay option is specified.
Finally since you are using python3 the 'from __future__  import print_function' is not needed.
This is only needed when you also want to run the script with python2.

The display has an atmel microcontroller which is intialized by the GPU (not the CPU) at boot.
Probably the 5V feed to the display has to be interrupted for it to properly initialize at reboot.

Hope you do not mind my comments. The script as such looks fine.

klint-k, 2019-01-17:
> @klint-k Some comments on your script.
> You have a problem when the UTC time is xx:59.
> Adding 1 minute will put the time in the past.
> Better to use Python a timedelta object of DELTA_MIN minutes and [...]

Thanks 
 I will try that.

klint-k, 2019-01-17:
change to   
```

def time_a():
    # Set RTC alarm 5 minutes from now
    # RTC is kept in UTC
    t = datetime.datetime.utcnow()
    a={}
    a['year'] = t.year
    a['month'] = t.month
    a['day'] = t.day
    if t.minute == 59:
        a['hour'] = t.hour + 1
        a['minute'] = (t.minute + DELTA_MIN) % 60
    else:
        a['hour'] = t.hour
        a['minute'] = (t.minute + DELTA_MIN) % 60
    a['second'] = t.second
    return a
```

closed 2019-09-15 by ChristopherRush