PiSupply/PiJuice #271: Compatability: 1.4 pijuice_base > python 3.4 - Request

donofriomark, 2019-01-17:
Looking for a "how to" guide to get 1.4 installed and working using python 3.4.
I have successfully completed 97% of the install manually, along with permissions, etc.
I can run the pijuice_cli and update both the board's mcu & json files. Pijuice.service loads and runs without issue, etc.
Specifically: the system_task:ext_halt_power_off:30 ... while the values in json are correct pijuice_sys.py is seemingly ignoring the comparison logic ? (starting at line 306).

Unfortunately, I need to use last jessie base build (2017-07-15) due to specific networking interface requirements. While I've successfully compiled/installed  python 3.5.6, 3.7.2 both the apt-get pijuice-base package & manual install from .deb continue to error out recognizing only the python 3.4.2. I've even changed all symlinks and removed pyth3.4 with no change to the exit error.
Any assistance would be greatly appreciated.

Thanks

tvoverbeek, 2019-01-17:
Assuming you are running pijuice_sys (the service) as user pijuice as specified in the pijuice.service file 
(https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-base/pijuice.service)
Line 301 in pijuice_sys.py (`sysJobTargets = subprocess.check_output(["sudo", "systemctl", "list-jobs"]).decode('utf-8'`) has to run with root privileges, otherwise the output does not contain the targets which are checked later.
This is ensured by line 1 in https://github.com/PiSupply/PiJuice/blob/master/Software/Source/data/020_pijuice-nopasswd.
Have you installed 020_pijuice-nopasswd in /etc/sudoers.d? See setup.py in the Source directory (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/setup.py)

donofriomark, 2019-01-17:
Yes, 020_ ... is in /etc/sudoers.d/  and ensured the 'pijuice' user is in groups: pi, sudo, i2c, pijuice.

shawaj, 2020-12-05:
@donofriomark did you need any further support on this at all? Or can we close the issue