PiSupply/PiJuice #276: PiJuice timesync query

satishgunjal, 2019-01-31:
My RPI is not connected to internet. Will PiJuice able to maintain the system clock during shutdowns?

As per your document 

> Sync RTC time at boot
When the Raspberry Pi shutsdown and then reboots you must copy the RTC time back to the system clock at boot and you can do this in /etc/rc.local with sudo hwclock -s.

What does it mean? Without internet connection after boot PiJuice wont be able to maintain the system clock out of the box??

tvoverbeek, 2019-01-31:
@satishgunjal 

> What does it mean? Without internet connection after boot PiJuice wont be able to maintain the system clock out of the box??

No, not without some software running at boot.
As long as the PiJuice itself has power its RTC will remain running, but the RPi system clock stands still between shutdown and the next boot.
To get the correct system time you have to copy the time in the PiJuice clock (RTC) to the system clock.
In PC-based linux systems this is done very early in the boot, since they have a built-in hardware clock.
The RPi does not and is depending on an external clock accessed via the I2C bus. The i2c bus is not yet available early in the boot, hence you have to do it later via e.g. /etc/rc.local

ChristopherRush, 2019-09-15:
Documentation here - https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-rtc

closed 2019-09-15 by ChristopherRush