PiSupply/PiJuice #278: Just started using PiJuice- Multiple Issues

satishgunjal, 2019-01-31:
##Issue:1
I want RPI to stat whenever power is connected to PIJuice. As per your comment in issue 240 I am unable to setup the wakeup on charge trigger level to 0%. I can give 1 instead of 0 but then also after saving settings i received below response.

"Settings saved
 Failed to communicate with PiJuice service."

I checked the pijuice service status its "desabled". 

##Issue:2
There is no program running on RPI 3B+. OS is Raspbian Stretch Lite still RPI is considerable warm!! 

##Issue:3
Able to start RPI by pressing SW1 but not able to shutdown the RPI if I press SW1 for 10 seconds!!

tvoverbeek, 2019-01-31:
All caused by i2c not working. Fix that first.

shawaj, 2019-02-01:
Yes you need to enable i2c. You call do this from the Raspberry Pi config menu. Click the menu icon at top left of desktop then on Preferences menu and find this tool and enable i2c

satishgunjal, 2019-02-01:
i am using Raspbian Stretch Lite OS. I can enable it from 'raspi-config'. But its never mentioned anywhere in your document to enable it before installing pijuice base version.

tvoverbeek, 2019-02-01:
@satishgunjal In the documentation it is not mentioned to enable i2c via raspi-config because the installation script of the pijuice-base package will enable it.
At that point you got the error reported in issue #275.
Again fix your Raspbian installation first, See my comment in issue #275 (https://github.com/PiSupply/PiJuice/issues/275#issuecomment-459483788)

GizmoEV, 2019-02-03:
@satishgunjal 
> "Settings saved
> Failed to communicate with PiJuice service."

I had the same issue installing on my RPi3B+. I don't remember where I found it but I ended up running `sudo systemctl enable pijuice.service` followed by `sudo service pijuice restart`. Now I don't have the errors any more. The only issue I still have is that the "System switch" setting doesn't stick through a power cycle. I need it on to run my 7" RPi display.

satishgunjal, 2019-02-04:
**I have formatted the SD card and installed Raspbian stretch lite. And performed below steps.**
1. sudo apt-get update
2. sudo apt-get upgrade (reboot after this)
3. sudo apt-get install python-dev
4. sudo apt-get install python-setuptools
5. sudo easy_install rpi.gpio (reboot after this)
6. sudo apt-get install pijuice-base
pijuice installation this time completed without any error. Can see pijuice service enables and running!.

**Issues are as below
(Using pijuice microusb port for power in)**
1. But battery level show s 0%
2. LED1 is constant Red (Since charging for sometime, battery shouldn't be < 15% then may be issue with battery??)
3. After pressing SW1 for 3 seconds rpi wont start. To start RPI i have connect power to RPI microusb

**How to troubleshoot it? Is my pijuice faulty??**

tvoverbeek, 2019-02-04:
@satishgunjal First steps 3,4,5 are not needed for pijuice-base installation.
If the battery is still low then have you read the important note in the main Readme?
```
Important note: If your PiJuice was supplied with a battery fitted, you will need to remove the little
plastic battery isolation tab. If you don't remove that plastic tab, your PiJuice will not charge up,
and will report 0% battery level.
```
I suspect the little isolation tab is still there.
But you are not alone, I had the same problem originally.

satishgunjal, 2019-02-04:
Actually I did read above notice and removed the plastic tab. My problem was battery contacts were not perfectly connected with board contacts. After removing and placing battery again issues resolved.

Please let me know, where my below configurations are correct.
1. I want RPI to keep time during reboot. rc.local file content.
   
```
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi

sudo hwclock -s
printf "RTC time copied to system clock"

exit 0
```

2. I want RPI to shutdown when battery is 10% and start again whenever mains power is back. ( 5V 2A adapter will always be connected to pijuice)
System task snap as below. 
[https://i.imgur.com/V0PdsOE.jpg](url)

3. Battery level fluctuates a lot. When connected to power status shows 90%. Once you remove the power in few minutes status shows 28%. RPI 3B+ headless without any program running or hardware connected to it last only for 1 hours max!!

4. If RPI is shutdown(SW1 button hold for 10 sec) then also RPI and pijuice red lights are on. is it correct behavior? wont it consume the battery?

5. Where can i see pijuice logs. I want to see whether pijuice started and shutdown as per my configuration.

closed 2019-10-23 by ChristopherRush