PiSupply/PiJuice #280: pijuice_cli - battery status oscillates a lot

cjdcordeiro, 2019-02-04:
Hi,

I'm testing my new pijuice hat with a pijuice solar panel (12W), on a RPi 3B+ with Raspbian Stretch lite installed. I'm measuring the battery status through the CLI and for some reason, **the percentage value jumps all over**. I've seen it going from 28% to 75% within seconds for example...I'm not sure how trustworthy this is and I wonder whether I'm misreading these values or the hardware is faulty.

Any hints?
Thanks

shawaj, 2019-02-04:
This is a known issue with the battery fuel gauge chip.

But later versions of the firmware for PiJuice (discussed here https://github.com/PiSupply/PiJuice/issues/249) have made improvements to this?

Thanks

cjdcordeiro, 2019-02-04:
thanks I'll follow that thread

closed 2019-02-04 by cjdcordeiro