PiSupply/PiJuice #281: System switch setting not persistent across power cycle

GizmoEV, 2019-02-05:
I have searched and looked through all open and closed issues but I don't see any thing related the issue I'm having. I have the current stable build of the PiJuice software and upgraded my PiJuice to the 1.2 firmware which came with the software download. I have a RPi 3B+ and the official Raspberry 7" display. I'm powering the display from the J3 connection. I set System switch to the 500mA setting. I can reboot and the display powers up just fine but if I shut down and then power backup the setting reverts to Off. I have to remote in, change the setting, and reboot to get the display powered again. It doesn't matter if I use the GUI or text interface to make the change or if I set it to the 2100mA setting, the setting doesn't stick through a power cycle. Is the setting written to a config file I could check? Any suggestions?

Thank you,

David

tvoverbeek, 2019-02-05:
@GizmoEV Confirmed. Shen you use the SYS_FUNC_HALT_POW_OFF or SYS_FUNC_SYS_OFF_HALT
the pijuice service sets the power switch to off (which is what you want, since also the external equipment has to be powered off).
However the previous state of the switch is not saved. When rebooting the gui/cli reads the current state of the switch (which is off) and displays it.
If you use the new external system halt function (when you do a shutdown programmatically) the power to the Pi via the 5V GPIO will be removed, but the VSYS remains (i.e. the screen remains powered). This is also not ideal.
Have to think how to solve this.

GizmoEV, 2019-02-06:
Thank you @tvoverbeek, at least I know I'm not doing something wrong. I know that powering the display on after the RPi boots doesn't work. I don't know where in the boot cycle it has to be up to be recognized but I was thinking that maybe there is a command line option which could be run early enough in the boot process to work? Should I just power the display from pins 2 or 4 for +5V and 6 for ground?

tvoverbeek, 2019-02-06:
@GizmoEV As far as I know the display is initialised very very early by the GPU even before the ARM side is running. So they need to be powered on simultaneously. I would power the display from the 5V GPIO pins. Then you do not have the problem with the SYS power switch. I have run the combo RPI+Display+PiJuice this way without problems.

GizmoEV, 2019-02-06:
Perfect, that is how I'll wire it up. Thank you.

closed 2019-02-06 by GizmoEV