PiSupply/PiJuice #285: Wakeup enabled not being set after restart despite using crontab triggered script

codesqueak, 2019-02-10:
I have installed the PiJuice CLI software on a new install of raspbian-stretch-lite (13/11/2018)

After shutdown, the RPI restarts automatically via a wakeup alarm as expected..

I have added the wakeup alarm script to enable repeated wakeups.  This does not appear to work as expected as the [ ] Wakeup enabled checkbox is now unchecked.

The script is enabled in crontab with:

`@reboot /usr/bin/python3 /home/pi/wakeup_enable.py >> /home/pi/debug.log`

Notes:

1. If I execute the script manually, the checkbox is enabled - the script appears to be good.
2. I added a 'print' to the pj.rtcAlarm.SetWakeupEnabled(True). This returns '{'error': 'NO_ERROR'}'
3. After restarting, the 'debug.log' file contains '{'error': 'NO_ERROR'}', i.e. is appears to work, but the checkbox is **not** enabled.

'

tvoverbeek, 2019-02-10:
I suppose you are having an internet connection on your RPi.
The cron @reboot jobs run very early in the boot sequence, before the network is up.
When the network comes up the timesyncd syncs the internet time to the system clock **and to the RTC**.
Since it uses the ds1339 module (which has a settable alarm from Linux) it resets the alarm.

What you can do is to include a delay at the start of /home/pi/wakeup_enable.py to be sure the network is up.

Alternatively you can set the eeprom address to 0x52. Then the ds1339 driver will not be loaded.
If you need the internet time sync add the following line to /boot/config.txt:
`dtoverlay=i2c-rtc,ds1307`.
This will load the ds1307 driver instead which has no alarm function. Hence the wakeup flag will not be reset by the timesyncd daemon.
You then also do not need wakeup_enable.py script since the flag will remain enabled.

codesqueak, 2019-02-17:
Yes, the device was connected to the network.  

I found putting in a delay wasn't reliable.  Even with very long delays, wakeup restart wasn't always enabled.

Changing the eeprom address worked without problem.

tvoverbeek, 2019-02-17:
@codesqueak Yes, the delay would be unreliable with the ds1339 driver loaded and the timesyncd daemon running.
Timesyncd synchronizes the time from the internet to the system and to the RTC (and clears the alarm flag in case of the ds1339) about every 15 minutes.
I am closing the issue. If you do not agree, please reopen it.

closed 2019-02-17 by tvoverbeek