PiSupply/PiJuice #286: What is ID of EEPROM?

rossirm, 2019-02-11:
Hi, I wonder what means ID EEPROM address of 0×50 or 0×52!? Is it the address for I2C interface or it means some address for anywhere else? If it means that this address is for I2C, why with command sudo i2cdetect -y 1 in Terminal there is no visualisation of this address when put PiJuice on Raspberry Pi 3B???

ryanteck, 2019-02-11:
Hi Rossirm,

You should see the Pijuice when running that command with a 14 showing at location 10,4 and UU at 60,8

The EEPROM is on the separate I2C bus for EEPROMs so will not show. For it to show you will have to enable the i2c0 bus and detect on that. However you shouldn't require to.

rossirm, 2019-02-11:
Thanks much! :) How can I enable this I2c0 bus? What do you mean under 'you shouldn't require to'!?

ryanteck, 2019-02-11:
Hi Rossirm,

You shouldn't usually need to enable the i2c0 bus as the EEPROM is automatically read by the Raspberry Pi OS and there's no benefit of enabling it to see the ID DIsplay as it won't functionally change anything.

There's information about EEPROMs on Raspberry Pi's at https://github.com/raspberrypi/hats, to enable i2c0 you need to add ```dtparam=i2c_vc=on``` to the /boot/config.txt file

rossirm, 2019-02-11:
Thanks so much! I only didn't understand why there are 2 i2c buses? 🤔

tvoverbeek, 2019-02-11:
Each HAT shuld have an eeprom at address 0x50 which can spevcify which gpios are used and also can contain a device-tree fragment.
In case of the PiJuice it contains a device-tree fragment which is the equivalent of
`dtoverlay=i2c-rtc, ds1339`.
The Pi scans address 0x50 on i2c bus 0. Normal i2c peripherals use i2c bus 1.
i2c bus 0 is also used by the GPU to control the Pi camera. Therefore it is not a good idea to access i2c bus 0 directly.
However what you can do is use bit-banged i2c communication to read or write the eeprom.
See e.g. https://github.com/raspberrypi/hats/blob/master/eepromutils/eepflash.sh which uses bitbanged i2c. The essential line is `dtoverlay i2c-gpio i2c_gpio_sda=0 i2c_gpio_scl=1`
This creates an i2c bus i2c-3 on the pins normally used for i2c-0 and connected to the i2c eeprom on a HAT.

closed 2019-02-11 by rossirm