PiSupply/PiJuice #293: pijuice-base : Depends: python3 (>= 3.5) but 3.4.2-2 is to be installed - revisited

StudioBeneden, 2019-02-26:
I'm having the exact same issue as #265 , I'm on 2017-07-05 jessie, and can't (easily) upgrade to stretch because of another install that's dependant on Jessie.. I've tried downgrading Python to 3.4.2 manually. python3 -V shows: Python 3.4.2, but the piJuice installer still refuses to run with the same warning. Hope there's a way around this? (Tried to add this as a comment to the closed issue #265 Sorry if that's bad practise..)

tvoverbeek, 2019-02-27:
Try to install the previous version of the packages (1.3.3) manually.
You can find them here:
https://github.com/PiSupply/PiJuice/blob/master/Software/Install/pijuice-base_1.3.3_all.deb
https://github.com/PiSupply/PiJuice/blob/master/Software/Install/pijuice-gui_1.3.3_all.deb
You probably will have to purge the partially installed 1.4 packages first.
Then install them manually with:
`sudo dpkg -i pijuice-base_1.3.3_all.deb`
and if you need the gui:
`sudo dpkg -i pijuice-gui_1.3.3_all.deb`
These still use python 2 and should work on jessie, although you do not get all the latest bells and whistles.

StudioBeneden, 2019-02-28:
Thank you for your help! This worked for me. Mind that I had to install urwid by hand too (`apt-get install python-urwid`) for the pijuice install to work. And for some reason downloading the .deb files with `wget` resulted in corrupt archives, so I had to copy them over via smb.

ChristopherRush, 2019-02-28:
I will update the documentation to reflect this.

tvoverbeek, 2019-02-28:
See also #294

StudioBeneden, 2019-02-28:
thanks again, closing the issue, hoping it helps other people with a similar situation!

closed 2019-02-28 by StudioBeneden