PiSupply/PiJuice #294: Unable to run pijuice_cli - missing module pijuice

Majkl578, 2019-02-27:
Trying to run pijuice_cli errors out with the message complaining about missing pijuice python module:
```
# pijuice_cli
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 15, in <module>
    from pijuice import PiJuice, pijuice_hard_functions, pijuice_sys_functions, pijuice_user_functions
ModuleNotFoundError: No module named 'pijuice'
```

Installed from Raspbian repository.
```
i# dpkg -l | grep pijuice
ii  pijuice-base                    1.4                          all          Basic support for Pi-Supply's PiJuice HAT
# cat /etc/debian_version
buster/sid

```

tvoverbeek, 2019-02-28:
Caused by running buster/sid instead of stretch.

You can try to copy /usr/lib/python3.5/dist-packages/pijuice.py to the python version dist-packages directory used by buster/sid (3.6 or 3.7?).
That should probably work.
The second error when running python3.5 (not finding smbus) is because the buster/sid python3-smbus only supports python3.6 and python3.7, not python3.5

Could you report back if moving the pijuice.py module works?
Good to know in preparation for when Raspbian switches to buster.

Majkl578, 2019-02-28:
(sorry, wrong button)

I'll try later today and report back.

Majkl578, 2019-02-28:
Actually got to it now.

Linking `/usr/lib/python3.5/dist-packages/pijuice.py` to `/usr/lib/python3.7/dist-packages/pijuice.py` made PiJuice CLI work. 👍 

Linking the module into `/usr/lib/python3/dist-packages` also works. Is there a reason why you install into vesion-specific dist-packages instead of python3 global, like other Debian python3-* packages do?
IMO you should provide appropriate package Depends/Breaks on (in)compatible python versions.

Thanks.

tvoverbeek, 2019-02-28:
@Majkl578  Thanks a lot for testing.
We use stdeb (https://pypi.org/project/stdeb/) to build the packages, It places pijuice.py in the dist-packages of the generating python version (For 1.4 this was 3.5).
Will see if I can modify the build procedure to put it in /usr/lib/python3/dist-packages instead to make it more future proof for all python 3.x versions.
This would have prevented the problems you had (and the problems we will get when Raspbian switches to buster)

shawaj, 2020-10-24:
@tvoverbeek anything we need to do here still or can we close?

tvoverbeek, 2020-10-24:
Nothing to do. The packages are also linked from /usr/lib/python3/dist-packages nowadays.

closed 2020-10-24 by tvoverbeek