PiSupply/PiJuice #298: Power Button to wake up and shut down the raspberry

mschuler, 2019-03-20:
According to the [https://github.com/PiSupply/PiJuice/blob/master/Hardware/README.md](Hardware README) i can short GPIO 5 & 6 to wake up the Raspberry Pi. Is there a comparable way to shut it down? I'd like to add a sort of on/off switch. Or is there another way to do that?

tvoverbeek, 2019-03-20:
Note GPIO pin 5 is used for the I2C clock (SCL). GPIO pin 6 is Ground.
The wakeup only works when the RPi is powered and in shut down state. Then the GPU (not the ARM CPU) is monitoring pin 5 and continues the boot when it is grounded. After the ARM CPU boots pin 5 is used as SCL if I2C is enabled (which is the case for the PiJuice).
To get the off function with a switch you need to monitor a different GPIO pin (when I2C is enabled) and when triggered execute a shutdown command.

LuvSachdevaRFID, 2019-08-07:
Is there any way I can bootup raspi without pushing the SW1 button.
I want to do it using the software.

tvoverbeek, 2019-08-07:
@LuvSachdevaRFID Try the following:
On the System Events tab enable 'wakeup on charge' with a trigger level of 0%.
As soon as the PiJuice gets power via its USB it should boot the Pi.

LuvSachdevaRFID, 2019-08-07:
> @LuvSachdevaRFID Try the following:
> On the System Events tab enable 'wakeup on charge' with a trigger level of 0%.
> As soon as the PiJuice gets power via its USB it should boot the Pi.

Thanks for your reply. I set up the wake-up charge in System tasks with trigger level 1%, But it didn't boot up.I have to manually press SW1 button :(

tvoverbeek, 2019-08-18:
@LuvSachdevaRFID See my comment here (https://github.com/PiSupply/PiJuice/issues/357#issuecomment-522326448) about how wakeup-on-charge works.

closed 2019-09-15 by ChristopherRush