PiSupply/PiJuice #299: Doesn't power on raspberry after 2-3 days power loss

robster0326, 2019-03-23:
Hi All!

In general pijuice works perfectly, but when i leave it power off for the weekend, and power it on after, it doesn't turn my Rpi on. Wakeup on charge is set ON. 
If i press the power on button, everything is fine, Rpi is booting. But if i leave it again powerloss for a few days, it doesn't power the Rpi on again :( If i leave it powerloss only for a day, it turns my Rpi on perfectly. 
It seems that if battery is empty, it doesn't turn Rpi on when its plugged in.

How can i eliminate this issue?

Thanks!

tvoverbeek, 2019-03-25:
@robster0326 How do you power off tje RPi for the weekend?
I am assuming you power the PiJuice via its own micro USB.
The 'wakeup on charge' setting is only copied to the PiJuice when it is shut down due to a 'low_charge',  'low_battery_voltage' or 'no_power' event which then triggers a SYS_FUNC_HALT or SYS_FUNC_HALT_POW_OFF.
If you execute the shutdown yourself, the 'wakeup on charge' setting is not copied to the PiJuice.
As soon as the PiJuice starts the RPi (e.g by a SW1 button press) the 'wakeup_on_charge' setting in the PiJuice MCU is cleared.
The value is not stored in non-volatile memory. So if the battery is completely empty, i.e. the PiJuice MCU is completely off (No slow weak blinking LEDs) then 'wakeup on charge' will not work.
Best to use SYS_FUNC_HALT_POW_OFF which turns off the 5V to the RPi. WIth the PiJuice in low power mode (weak blinking low frequency LED) it should last several days.

robster0326, 2019-03-25:
Hi Ton!

Actually i use my RPi in my car, so the pijuice is powered through its own microUSB when the car is ignited. When i turn off the car, I expect the pijuice to gently turn off the RPi not to damage the SD Card. It worked a while, but now it does not turn on the RPI when I ignite the car. I set the No Power system event to SYS_FUNC_HALT_POW_OFF and set the Wakeup on charge=trigger level 1%.

If I understand correctly, if the battery is full empty, it will not wake up my RPI on charge?

tvoverbeek, 2019-03-26:
@robster0326 Yes, if the battery is really empty (= PiJuice MCU has to do a cold boot) the wakeup on charge does not work. I suppose the USB cable remains connected when the car ignition is off. Anything else connected which could drain the battery when the ignition is off?

robster0326, 2019-03-26:
@tvoverbeek No, just the RPi is connected... after a day or two, it drains to battery and it won't wakeup RPi :( Yes, the USB cable remains connected to my car's power outlets at the moment. Maybe I should power Pijuice from the car's battery directly. There is an external HDD connected to the RPi, but that also turns off when Pijuice shuts RPi down.

tvoverbeek, 2019-03-26:
@robster0326 As a test could you disconnect the USB cable from the car's power outlet over the weekend and check if it also does not boot when you reconnect the USB cable after 2-3 days?
Otherwise connecting it directly to the 12V car battery could work, but since it then remains powered how do you (re)start the RPi ?

robster0326, 2019-03-26:
@tvoverbeek Yes, i will give it a try. That's a good question :) In this case, RPi would remain on...

closed 2019-09-15 by ChristopherRush