PiSupply/PiJuice #301: VSYS output usage question

smallhacks, 2019-03-25:
Hi,

I read in the documentation that J3 pin (VSYS) can be used to provide power to external sources up to 500mA or 2100mA.
Could it be used to power and control a 5V cooling fan ?

Thank you!

tvoverbeek, 2019-03-25:
VSYS is in principle the battery voltage (nominal 3.7V, varies between ca 3 and 4.2V depending on charge level). So if your cooling fan can run with these voltages and current limits then it should work.

smallhacks, 2019-03-25:
Thank you for your response.
I tried with a mini fan and it seems to work pretty well!

francesco-vannini, 2019-03-25:
I'd suggest to add this to the hardware README.md as it is not that obvious.

tvoverbeek, 2019-03-26:
Hardware README.md updated. Closing the issue.

closed 2019-03-26 by tvoverbeek