PiSupply/PiJuice #304: Possible software defect in pyjuice.py:SetTime()

WillBaldwin, 2019-03-26:
### Possible software defect in pyjuice.py:SetTime() 
 
I believe I discovered a software defect in **pyjuice.py:SetTime()** at **line 632** where both the if and the elseif are compared to the same exact value to determine if d[8] could contain 0x02 or 0x01.

Code:
                if 'daylightsaving' in dt:
                if dt['daylightsaving'] == _**'SUB1H'**_:
                d[8] |= 2
                elif dt['daylightsaving'] == **_'SUB1H'_**:
                d[8] |= 1

tvoverbeek, 2019-03-26:
You have an old version of pijuice.py.
Current version has the following code:
```
        if 'daylightsaving' in dt:
            if dt['daylightsaving'] == 'SUB1H':
                d[8] |= 2
            elif dt['daylightsaving'] == 'SUB1H':
                d[8] |= 1
```
Moreover in the old code it will assign 0x02 since the else if (elif) will not be executed.

tvoverbeek, 2019-03-26:
@WillBaldwin Yes this is wrong. It is already fixed on the develop branch.
There the code is:
```
        if 'daylightsaving' in dt:
            if dt['daylightsaving'] == 'SUB1H':
                d[8] |= 2
            elif dt['daylightsaving'] == 'ADD1H':
                d[8] |= 1
```
When using Raspbian/Linux this is not a problem since the RTC is always kept in UTC which does not have daylight saving time.
I do not know what the convention is in Windows IOT.

closed 2019-09-16 by ChristopherRush