PiSupply/PiJuice #305: Impossible to set battery profile, always errors out

Majkl578, 2019-03-29:
I was trying to set custom battery profile - after setting it to Custom I started getting continous errors upon doing Apply changes.

Right now I can't even open Battery profile menu, it always crashes:
```
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 1808, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 719, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 393, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 493, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 403, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 503, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 592, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1587, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2269, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 621, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 985, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 621, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 535, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 463, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 264, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 294, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1709, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 932, in __init__
    self.refresh()
  File "/usr/bin/pijuice_cli.py", line 999, in refresh
    self.main()
  File "/usr/bin/pijuice_cli.py", line 947, in main
    urwid.IntEdit("Capacity [mAh]: ", default=self.profile_data['capacity']),
TypeError: string indices must be integers
```

Tried deleting /var/lib/pijuice/pijuice_config.JSON, no help. Tried resetting PiJuice with SW1 + SW2 for 20 seconds, no help.

Status reports "Fault: battery profile invalid".

At this state PiJuice is unusable. Also this software is very buggy. :(

Majkl578, 2019-03-29:
After manually calling python library, I think I got it working (for 12000 mAh official PiJuice battery):

```
>>> pj = PiJuice()
>>> config = PiJuiceConfig(pj.interface)
>>> profile = {}
>>> profile["capacity"] = 12000
>>> profile["chargeCurrent"] = 2400
>>> profile["terminationCurrent"] = 600
>>> profile["regulationVoltage"] = 4180
>>> profile["cutoffVoltage"] = 3000
>>> profile["tempCold"] = 1
>>> profile["tempCool"] = 7
>>> profile["tempWarm"] = 45
>>> profile["tempHot"] = 60
>>> profile["ntcB"] = 3380
>>> profile["ntcResistance"] = 10000
>>> profile
{'cutoffVoltage': 3000, 'terminationCurrent': 600, 'chargeCurrent': 2400, 'capacity': 12000, 'ntcResistance': 10000, 'tempCool': 7, 'tem          pCold': 1, 'regulationVoltage': 4180, 'ntcB': 3380, 'tempWarm': 45, 'tempHot': 60}
>>> config.GetBatteryProfile()
{'data': 'INVALID', 'error': 'NO_ERROR'}
>>> config.SetCustomBatteryProfile(profile)
[224, 46, 24, 11, 34, 150, 1, 7, 45, 60, 52, 13, 232, 3]
{'error': 'NO_ERROR'}
>>> config.GetBatteryProfile()
{'data': {'cutoffVoltage': 3000, 'terminationCurrent': 600, 'chargeCurrent': 2350, 'capacity': 12000, 'ntcResistance': 10000, 'tempCool'          : 7, 'tempCold': 1, 'regulationVoltage': 4180, 'ntcB': 3380, 'tempWarm': 45, 'tempHot': 60}, 'error': 'NO_ERROR'}
>>> config.SetCustomBatteryProfile(profile)
[224, 46, 24, 11, 34, 150, 1, 7, 45, 60, 52, 13, 232, 3]
{'error': 'NO_ERROR'}
>>> config.SetBatteryProfile("CUSTOM")
{'error': 'NO_ERROR'}
```

tvoverbeek, 2019-03-30:
@Majkl578 Yes, this is a serious bug in pijuice_cli. When you select the custom profile and apply settings before you have specified the custom profile data then you end up in this situation where you cannot recover.
By using the API (pijuice.py) to define a valid custom profile you could recover.

Working on a fix so other people do not end up in this situation.

Before all the other battery sizes were available from PiSupply we did not need the custom profile.
Hence this error was not noticed before.
In the next software release the profiles for all the PiSupply batteries will be predefined in the software.

Thanks for reporting this serious bug.

PS Your website https://majkl.me is not accessible from Chrome (NET::ERR_CERT_DATE_INVALID)

ThomasDalmayrac, 2019-04-01:
Hi @Majkl578, can you describe for beginners how to manually call the pyhton library ? :)
Thanks for the solution, I'm also stuck with the 12000mA.

Majkl578, 2019-04-01:
@ThomasDalmayrac Run `python3` (interactive) and then reapply commands from my [previous comment](https://github.com/PiSupply/PiJuice/issues/305#issuecomment-478172534) (every `>>>` is a separate statement).
In my case the config is for 12000 mAh battery, if you need different, don't forget to adjust values accordingly.

@tvoverbeek Thanks. (Also, cert fixed. 😂)

ThomasDalmayrac, 2019-04-01:
@Majkl578 thanks for the fast answer ! In which folder should I be in ?
It's probably an idiot question, but I've got this result :
```
pi@raspberrypi:~ $ python3
Python 3.5.3 (default, Sep 27 2018, 17:25:39)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> pj = PiJuice()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'PiJuice' is not defined
```

Majkl578, 2019-04-01:
Folder is irrelevant. I see I forgot to include initial import statement, so _0th_ statement is:
```
from pijuice import PiJuice
```

tvoverbeek, 2019-04-01:
One addition:
The line `config = PiJuiceConfig(pj.interface)` should be
`config = pj.config`

ThomasDalmayrac, 2019-04-02:
Yes ! Thanks @tvoverbeek for the config fix. Thanks @Majkl578, it works like a charm :)
So the entire code that wiorked for me is : 
```
>>> from pijuice import PiJuice
>>> pj = PiJuice()
>>> config = pj.config
>>> profile = {}
>>> profile["capacity"] = 12000
>>> profile["chargeCurrent"] = 2400
>>> profile["terminationCurrent"] = 600
>>> profile["regulationVoltage"] = 4180
>>> profile["cutoffVoltage"] = 3000
>>> profile["tempCold"] = 1
>>> profile["tempCool"] = 7
>>> profile["tempWarm"] = 45
>>> profile["tempHot"] = 60
>>> profile["ntcB"] = 3380
>>> profile["ntcResistance"] = 10000
>>> profile
{'cutoffVoltage': 3000, 'terminationCurrent': 600, 'chargeCurrent': 2400, 'capacity': 12000, 'ntcResistance': 10000, 'tempCool': 7, 'tem          pCold': 1, 'regulationVoltage': 4180, 'ntcB': 3380, 'tempWarm': 45, 'tempHot': 60}
>>> config.GetBatteryProfile()
{'data': 'INVALID', 'error': 'NO_ERROR'}
>>> config.SetCustomBatteryProfile(profile)
[224, 46, 24, 11, 34, 150, 1, 7, 45, 60, 52, 13, 232, 3]
{'error': 'NO_ERROR'}
>>> config.GetBatteryProfile()
{'data': {'cutoffVoltage': 3000, 'terminationCurrent': 600, 'chargeCurrent': 2350, 'capacity': 12000, 'ntcResistance': 10000, 'tempCool'          : 7, 'tempCold': 1, 'regulationVoltage': 4180, 'ntcB': 3380, 'tempWarm': 45, 'tempHot': 60}, 'error': 'NO_ERROR'}
>>> config.SetCustomBatteryProfile(profile)
[224, 46, 24, 11, 34, 150, 1, 7, 45, 60, 52, 13, 232, 3]
{'error': 'NO_ERROR'}
>>> config.SetBatteryProfile("CUSTOM")
{'error': 'NO_ERROR'}
```

IX9, 2019-07-25:
I'm having the same issue (I believe) with the 12000mah battery.

When using pijucie_cli v1.5 and selecting "Battery Profile" it crashes with the following;

Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2007, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 278, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 376, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 682, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 719, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 393, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 493, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 403, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 503, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 592, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1587, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2269, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 621, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 985, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 621, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 535, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 463, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 264, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 294, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1907, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 1003, in __init__
    self.refresh()
  File "/usr/bin/pijuice_cli.py", line 1123, in refresh
    self._read_chemistry()
  File "/usr/bin/pijuice_cli.py", line 1181, in _read_chemistry
    self.chemistries_idx = self.CHEMISTRY_OPTIONS.index(self.ext_profile_data['chemistry'])
ValueError: 'UNKNOWN' is not in list

tvoverbeek, 2019-07-25:
In 1.5 the 12000 mAh battery is one of the predefined profiles.
To recover try the following:
- Start pijuice_cli and select General
- At the bottom of General Settings select 'Reset to default'
- Confirm. During the reset the Leds will flash red/blue
- After the reset select 'Battery profile'
- Battery settings should now say 'Status: Default profile'
  and below 'Profile: BP7x_1820' as selected.
- Just press enter and you get a 'Select battery profile' screen.
- Scroll down to PJLIPO_12000 and select it.
- Then back to 'Battery settings', scroll down and select 'Apply settings'

That should hopefully solve it.

IX9, 2019-07-25:
Thank you for the quick reply, that did solve the issue!

tvoverbeek, 2019-07-25:
Glad it worked. Closing the issue.

closed 2019-07-25 by tvoverbeek