PiSupply/PiJuice #308: powering pijuice from usb and solar is this ok?

giff1000, 2019-04-08:
I wonder is someone could help me with the below:

is it possible to power the pijuice from 2 different power supplies - the reason I ask is because currently I have a solar panel charging a voltaic powerbank which then plugs into the micro usb on the pijuice, however this is a bit of a waste because there is more power available than the powerbank/pijuice can use. So the idea would be to split the power coming in from the panel to power the pijuice directly and also charge the powerbank. the other advantage to this is that it allows the powerbank to charge in low sunlight rather than just passing through to the pijuice board.

Any help would be greatly appreciated

tvoverbeek, 2019-04-08:
Short answer: No
I am assuming you want the powerbank output connected to the PiJuice USB and also connect the solar panel output directly to the PiJuice. The only other input available is J4 (See hardware README).
However the J4+ is connected to the VBUS on the USB connector.
So the two inputs are not independent.
See also the hardware schematic in Documentation.

If this is not the setup you have in mind, please let us know.

shawaj, 2019-04-09:
You could connect one to the Pi instead perhaps?

tvoverbeek, 2019-04-09:
Yes, that is a possibility. Connect the power-bank 5V output to the Pi USB and the solar panel either to J4 or the PiJuice USB.
The PiJuice input has to be between 4.5 and 10V
Note the 'Input precedence' setting in the HAT configuration which determines which input is used to charge the PiJuice own battery if both inputs are present.

giff1000, 2019-04-11:
Brilliant thanks v much

giff1000, 2019-04-11:
sorry just a minor clarification; the supply from solar panel is converted down to 6v, I presume I cannot connect this to the pi juice usb, only the j4 pad

tvoverbeek, 2019-04-11:
@giff1000 You can connect the 6V solar panel output to either the PiJuice usb or to J4.

closed 2019-09-16 by ChristopherRush