PiSupply/PiJuice #309: PiJuice powerdown on power removal

geoffr98, 2019-04-09:
I'm trying to set-up the PiJuice to power off the Pi when power is removed, but I haven't been able to get it to work - How can I set-up the PiJuice to:

1. Power the Pi on when power is applied to the PiJuice J4
2. Power down the Pi on when power is Removed from the PiJuice J4
3. Still be able to power up/down using SW1
4. Power down the Pi fully when it is shutdown from within the terminal

tvoverbeek, 2019-04-10:
@geoffr98 
On the 'System Task' tab:
- Enable 'System task enabled'
- Enable 'Wakeup on charge' with a low trigger level, e.g. 0%
- Enable 'Software Halt Power off' with a delay period of at least 20 seconds

On the 'System Events' tab
- Enable 'No power' and assign the function 'SYS_FUNC_HALT_POW_OFF'

On the 'Buttons' tab of the HAT configuration for SW1
- SINGLE_PRESS:  HARD_FUNC_POWER_ON, Parameter 800
- LONG_PRESS1:  SYS_FUNC_HALT_POW_OFF, Parameter 5000 (= 5sec press)

This will shut down the Pi when external power is removed and after 60 seconds remove the 5V to the Pi.
When external power is applied the Pi will boot

SW1 will still power the Pi off and on.
After you shut down the Pi with SW1 you can **not** boot the Pi by cycling the external power.

The 'Software Halt Power off' setting takes care of item 4. When you shut down the Pi from the terminal or desktop the 5V power to the Pi will be removed after 20 sec.
Also here cycling the external power will **not** boot the Pi.

geoffr98, 2019-04-10:
Thanks - that works - What I had tried cause it to shudown every time (using the power levels on the battery for shudown - I missed the "'No power' and assign the function 'SYS_FUNC_HALT_POW_OFF'"

I don't have "Enable 'Software Halt Power off' with a delay period of at least 20 seconds" .. I think this is set by default in the latest 1.2 firmware.

tvoverbeek, 2019-04-11:
@geoffr98 It is in the 1.2 firmware and when you update the pijuice-base and pijuice-gui pacakges to the current version (1.4) you will have the  "Enable 'Software Halt Power off'  option on the system task tab.
Closing the issue. Reopen it if you do not agree.

closed 2019-04-11 by tvoverbeek