PiSupply/PiJuice #311: Maximum charge level at 85%

richardquint, 2019-05-05:
I had not used my RaspberryPi with a PiJuice hat for over a month.  I plugged in power directly to the PiJuice's microUSB socket (an Adafruit adapter) but even after continuously charging for several hours the battery level holds steady at 85%.  I tried discharging the battery by letting it run down to 5% and then recharging, but it still maxs out at 85%.  Could this simply be a sign of a failing battery?  I can swap the battery out, but it involves doing some disassembly, so I want to check before going there.

shawaj, 2019-05-05:
Do you unit what version firmware you are running?

richardquint, 2019-05-05:
It's 1.3
[image]

richardquint, 2019-05-05:
Looks like I was premature.  After waiting five extra hours and several reboots, the level has increased to 93%.  Perhaps it needed extra charging time after its hiatus.  I'll let it charge even longer.

shawaj, 2019-05-05:
Ok let us know of any developments

tvoverbeek, 2019-05-06:
@richardquint Note there is a later 1.3 beta firmware (2019-02-28) on https://github.com/PiSupply/PiJuice/tree/develop/Firmware/Beta
Early versions of the 1.3 Beta 'lost' the battery level after the firwmare was restarted (i.e after the battery went to low or was removed).
I do not remember if this was already fixed in the 2019-01-15 version

richardquint, 2019-05-06:
Thanks.  I'll update my firmware and see what happens.

flobernd, 2019-07-20:
I have the same problem. Got the Pi running for about 10 hours now and the battery level does not seem to pass 82%. Maybe a temperature iissue? (54° C).

flobernd, 2019-07-21:
@shawaj It's another 12 hours now and the charge level is still at 82%. Temperature is about 50°C after I slightly moved the Pi to a cooler place.

Is there anything I can do/test?

tvoverbeek, 2019-07-21:
@flobernd I am assuming you are using the BP7X battery.
For the BP7X the 'Warm temperature' setting is 45 ºC and  the 'Hot temperature' is 59 ºC.
From the hardware README:
```
Warm temperature. Defines temperature threshold according to JEITA standard above which the
battery regulation voltage is reduced by 140mV from the programmed regulation voltage. This
threshold should be set above cool temperature.

Hot temperature. Defines temperature threshold according to JEITA standard above which charging
is suspended. This threshold should be set above warm temperature.
```
My guess is you are above the reduced regulation voltage and only trickle charge is occurring.
The BP7X is also heated by the power dissipation of the HAT (the voltage regulators) and the RPi itself.
I am assuming the temperature sense is set to AUTO or NTC.
Can you try to charge for several hours with the Pi completely off (no red power led)?

You could also try to move the PiJuice HAT off the RPi to lower its temperature.
There are not that many connections needed:
+5V on pin 1
+3.3V on pin 18 (only if you use the HAT EEPROM)
pins 27 and 28 for the HAT EEPROM (not needed if you do not use it)
pins 4 and 6 for I2C
and e.g. pin 10 for ground.

Alternatively, if you have an external battery with NTC you could try to connect that to the HAT instead of the BP7X. Of course you will need to choose the correct battery profile for the external battery.

flobernd, 2019-07-21:
@tvoverbeek Thanks for your reply. Your assumtions are correct. I just put an old fan in front of the Pi (reported battery temperature is 24°C now) and it successfully charged up to 96%. Guess I have to mount a permanent fan to my Pi.

flobernd, 2019-07-21:
Is it normal, that the battery slowly discharges even if the Pi is permanently powered?

tvoverbeek, 2019-07-23:
Is charging enabled on the general HAT Config tab in the GUI?
If it is not, the battery will slowly discharge. Other symptoms then are LED1 not flashing, but solid green when charge > 50% and Battery Status NORMAL instead of CHARGING_FROM_5V_IO.

flobernd, 2019-07-23:
@tvoverbeek Thanks for reply again. Charging is enabled and the LED is blinking green/blue. After I mounted a fan to the Pi, the battery temp sensor constantly shows 24°C (seems not accurate, but whatever ...) and the battery finally managed to charge up to 97% (this is still the maximum - it won't charge further). But as soon as I disable the fan, the charge level slowly decreases (was on 91% after ~12 hours). Not sure if this is normal behavior caused by the "high" temperature or if my device has some hardware problems.

tvoverbeek, 2019-07-23:
What is the battery temperature and voltage?
If the temperature is above 45 and the battery voltage > about 4.01V then it will only trickle charge, which is probably not enough to compensate the PiJuice own consumption.
Similar with the fan with temperature below 45 but battery voltage > 4.18V only trickle charge.
My own PiJuice (now mounted on Pi4B) battery temperature is 47 and voltage 4.154 and charge level is stuck at 97.
When I shut down the Pi and  power the PiJuice it slowly charges till full (Led solid blue instead of flashing green/blue)

flobernd, 2019-07-23:
@tvoverbeek Temperature is at 24°C (at least thats what the sensor inside the PiJuice reports) and voltage at 4.157V. That's pretty close to the tickle charge limit so that sounds plausible.

> which is probably not enough to compensate the PiJuice own consumption

Doesn't the PiJuice get its power directly from the miniUSB? Why does it have to draw it from the battery?

closed 2019-09-16 by ChristopherRush