PiSupply/PiJuice #313: Feature: compatibility with Nebra AnyBeam?

Leobaillard, 2019-05-13:
Hi there!

I recently backed PiSupply and Nebra's Kickstarter campaign for the Nebra AnyBeam projector. I'm most interested in the Raspberry Pi HAT and I was wondering if it could be compatible with our PiJuice boards?

I think there would be mainly two aspects to this: power and i2c availability. I wouldn't mind using a bigger battery for a sufficient power output but the main question is to know if the max power envelope is enough for the Pi and the projector.

Thanks again for your great products!!

ryanteck, 2019-05-13:
It'll be something that will require testing to get some accurate results.

With a Pi Zero or original A+ it might be fine with the normal battery that is included. The projector draws approximately 1-1.5A with the Zero drawing around 0.2-0.5A. 

However with a  Pi 3 this might not work with the Motorola battery as the Pi itself can draw over 1A. 

@shawaj when my Anybeam HAT arrives I'll test it with the bench PSU for power measurements and with the Pijuice.

Leobaillard, 2019-05-13:
Nice, good to hear! I'm up to do some testing as well when I'll get mine, but I suspect that you will all have done that way before me :) Anyway, great news, we'll have something to investigate!

shawaj, 2019-05-13:
@ryanteck I will send you one tomorrow.

Also - the other thing is that the battery backup "dumb" features should work fine - but in some scenarios the I2C will not work as the DPI interface over the GPIO pins will use some of the I2C pins in the highest quality scenarios. Just something to be aware of.

tvoverbeek, 2019-05-13:
If you use RGB666 on the Anybeam HAT (mode 4) you can use software i2c on GPIOs 26 and 27.
See dtoverlay=i2c-gpio in /boot/overlays/README.
This bit-banged I2C bus shows up as bus 3 (/dev/i2c-3).
You will need to change the PiJuice code (pijuice.py, pijuice_tray.py, pijuice_bui.py, pijuice_cli.py)
to use bus 3: 'pijuice=PiJuice(1,0x14)' -> 'pijuice=PiJuice(3,0x14)'

shawaj, 2019-05-13:
@tvoverbeek thanks for that!

ryanteck, 2019-05-17:
Out of the box it works perfectly fine and actually seems to work with a Pi 3B+ however I would recommend using a bigger battery to prevent any issues.

However the shorter posts must be used instead of the taller posts that will be included.

shawaj, 2020-02-22:
@ryanteck thanks for testing.

@ChristopherRush do you have an AnyBeam HAT to hand? Reckon you could document the process for this at some point (perhaps on the AnyBeam GitHub https://github.com/NebraLtd/AnyBeam ) and then close the issue?

ChristopherRush, 2020-02-22:
I don't have a Anybeam HAT, I think only Ryan does. I only have the
standard one in the case.

shawaj, 2020-02-22:
@ChristopherRush aha, fair enough.

@ryanteck do you still need the hat currently? Could we send to @ChristopherRush ?

ryanteck, 2020-02-22:
I don't currently need it so can send it to Chris, will do on Monday

shawaj, 2020-02-22:
Thanks @ryanteck

ChristopherRush, 2020-04-20:
I'm struggling to get this to work. I added everything for "Mode 4" including `dtoverlay=i2c-gpio,i2c_gpio_delay_us=1,i2c_gpio_sda=26,i2c_gpio_scl=27` and the bus is there when I run `i2cdetect -l`. I changed PiJuice address in the following file: pijuice.py, pijuice_sys.py, pijuice_tray.py, pijuice_gui.py, pijuice_cli.py but I still receive communication error. When I run `i2cdetect -y 3` it is super slow and the address does not appear. Is there something i'm missing?

ChristopherRush, 2020-04-20:
Forgot to add that the service is running fine.
[image]

tvoverbeek, 2020-04-20:
@ChristopherRush Which Pi? On a Pi4 you get a different bus number (7 in my case) for the first i2c-gpio.
Just to state the obvious : GPIO26 = pin 37 and GPIO27 = pin 13.
I am assuming you do not have the PiJuice connected directly to the 40-pin header.
Also why i2c_gpio_delay_us=1 instead of the default 2?
Have you looked on a scope to see the actual clock frequency?

tvoverbeek, 2020-04-20:
@ChristopherRush Came to think of something.
The normal i2c bus (i2c-1) uses the built-in pull-up resistors on GPIO2 and 3.
The PiJuice has no built-in pull-up resistors.
So you need external pull-up resistors on your software i2c bus.
The missing pull-up resistors also explains the slowness on your i2c-detect.

ChristopherRush, 2020-04-21:
@tvoverbeek Added pullup resistors and that resolved the speed issue but still no address shows up. I'm using a 3B+, `i2cdetect -l` shows bus 3 created. For the time being I have removed the Anybeam HAT so PiJuice is stacked on top with pullup resistors in place.

ChristopherRush, 2020-04-21:
@tvoverbeek Ok, adding pullup resistors resolved it as well as `Just to state the obvious : GPIO26 = pin 37 and GPIO27 = pin 13` >< . Thanks for your guidance.

Leobaillard, 2020-10-10:
Hi! Thanks all for your much appreciated testing! As the shipping date nears for our AnyBeam HATs, I wanted to know if your latest testing was conclusive and if we could work on some sort of procedure or doc? I'd be happy to help in any way!

shawaj, 2020-10-10:
@ChristopherRush did you get anywhere with documenting this all?