PiSupply/PiJuice #314: Instant Shutdown on AC Power unplug

effort10, 2019-05-14:
Good Day,

Just a quick question, i have about 3 usb devices plugged into my pi and when I remove the power the pi/ups shuts down in about 5 seconds as I think it cant handle the power draw.

Would a bigger battery sort this issue out?

Thank you

shawaj, 2019-05-14:
With a bigger battery such as the 12000mAh you can output more amperage to the Pi and therefore it should indeed help

effort10, 2019-05-14:
> With a bigger battery such as the 12000mAh you can output more amperage to the Pi and therefore it should indeed help

Thank you. Will try this  option out :)

ChristopherRush, 2019-09-16:
We now stock larger batteries here - https://uk.pi-supply.com/search?type=product&q=pijuice+battery

closed 2019-09-16 by ChristopherRush