PiSupply/PiJuice #317: Pijuice Android things

riyal123, 2019-05-27:
Can we use Pijuice with Android Things and also get access to RTC?

tvoverbeek, 2019-05-27:
@riyal123 In principle yes.
But you need to provide a driver for the PiJuice.
The PiJuice only uses I2C communication for getting info and changing settings. Also the RTC is accessible via I2C.
I2C is supported in the Peripheral I/O. See https://developer.android.com/things/sdk/pio/i2c.html
You will have to extract the communication details from the existing Python pijuice.py

riyal123, 2019-05-28:
So for using Pijuice just as a battery backup for the Rasberry Pi do I need
to provide a driver?. Isn't it a plug and play for battery??

tvoverbeek, 2019-05-28:
@riyal123 Out of the box with the BP7X battery it should work fine as battery backup.
But you asked about RTC access. That probably will nt work out of the box with Android Things.
In any case I still recommend you check the battery parameters and other settings using either pijuice_cli or pijuice_gui on a standard Raspbian install.

riyal123, 2019-06-18:
I  am using the following hat for Rasberry PI 3B running Android things
 https://www.waveshare.com/li-ion-battery-hat.htm
So the use case is when the power goes off, it gets switched to battery power and fluctuation causes the Rasberry PI to restart. So is this the case of pijuice also?
The same restart happens when the power is back also.
Thanks in advance

shawaj, 2019-06-18:
No the PiJuice would not automatically restart in this scenario. This is what it is designed to cope with

closed 2019-09-16 by ChristopherRush