PiSupply/PiJuice #318: Any update on official release of v1.3?

dikodahan, 2019-05-29:
Hi,
Seems like FW v1.3 has been out for quite a while with a few notable fixes but still listed in beta mode. Can you please share an official release status for this version?

tvoverbeek, 2019-05-29:
@dikodahan I have been working as a volunteer to update the software for the 1.3 firmware.
Current status is pijuice.py, pijuice_cli and pijuice_gui are adapted for the 1.3 firmware.
What still needs to be done is testing and adapting the updated software to not conflict with the current 1.2 firmware.
When that is done the Raspbian packages will be produced (probably by me) and uploaded by PiSupply (@shawaj).
If you want you can download the needed python scripts and supporting executables from the development branch.

tvoverbeek, 2019-07-26:
Closing issue, since 1.3 firmware is in the 1.5 pijuice-base package

closed 2019-07-26 by tvoverbeek