PiSupply/PiJuice #319: Deep sleep power consumption?

gulbaek, 2019-05-30:
I'm building a setup with a e-ink monitor and a raspberry pi and looking for a solution where I can run it on a battery.

The system only needs to be active for a 1-2 min to update the e-ink  and then go back to some deepsleep mode for 24 hours and start up again.

So I would like to know what the power consumption is for the pijuice in deep sleep mode?

tvoverbeek, 2019-05-30:
I do not have actual figures. But when I leave my Pi3b+/PiJuice combination powered off (= PiJuice in low power mode, slow faint blinking LED) for over 2 weeks starting from 100% charged, then at power on the battery is still > 70%. An other datapoint: people using a powerbank to power the PiJuice/Pi combo have problems because the power consumption in low power mode is too low and the powerbank shuts down.
Suggest to just try your scenario for a week or so to get the figures you need.

shawaj, 2019-05-30:
I think @ryanteck might be trying this at some point soon

gulbaek, 2019-05-30:
@tvoverbeek Thanks for the info. I'm thinking about using the PiJuice 12000mAh Battery, instead of a powerbank. Hoping that it can handle the low power consumption and not shutdown like a normal powerbank.

tvoverbeek, 2019-05-31:
@gulbaek The 12000mAh battery will not shutdown. It replaces the BP7X and will have to be connected to the PiJuice via the screw terminal. Note you will have to find a place for it separate from the PiJuice/Pi combination. It is much larger than the BP7X.
For your scenario (one update/day of an e-ink display) it will hold very very long.
If you do not need internet connectivity, shut down WiFi and Bluetooth and also the HDMI to minimize power consumption of the Pi.

closed 2019-09-16 by ChristopherRush