PiSupply/PiJuice #320: Error on boot

fencer, 2019-06-02:
I've notice these errors in the Log:
`Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]: Traceback (most recent call last):
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]:   File "/usr/bin/pijuice_sys.py", line 280, in <module>
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]:     main()
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]:   File "/usr/bin/pijuice_sys.py", line 197, in main
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]:     reload_settings()
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]:   File "/usr/bin/pijuice_sys.py", line 166, in reload_settings
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]:     config_dict = json.load(outputConfig)
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]:   File "/usr/lib/python2.7/json/__init__.py", line 291, in load
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]:     **kw)
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]:   File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]:     return _default_decoder.decode(s)
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]:   File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]:     obj, end = self.raw_decode(s, idx=_w(s, 0).end())
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]:   File "/usr/lib/python2.7/json/decoder.py", line 380, in raw_decode
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]:     obj, end = self.scan_once(s, idx)
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[570]: ValueError: Expecting , delimiter: line 8 column 2 (char 114)
Jun  2 14:53:55 WetPitsVQ systemd[1]: pijuice.service: Main process exited, code=exited, status=1/FAILURE
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]: Traceback (most recent call last):
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]:   File "/usr/bin/pijuice_sys.py", line 280, in <module>
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]:     main()
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]:   File "/usr/bin/pijuice_sys.py", line 197, in main
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]:     reload_settings()
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]:   File "/usr/bin/pijuice_sys.py", line 166, in reload_settings
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]:     config_dict = json.load(outputConfig)
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]:   File "/usr/lib/python2.7/json/__init__.py", line 291, in load
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]:     **kw)
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]:   File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]:     return _default_decoder.decode(s)
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]:   File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]:     obj, end = self.raw_decode(s, idx=_w(s, 0).end())
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]:   File "/usr/lib/python2.7/json/decoder.py", line 380, in raw_decode
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]:     obj, end = self.scan_once(s, idx)
Jun  2 14:53:55 WetPitsVQ pijuice_sys.py[630]: ValueError: Expecting , delimiter: line 8 column 2 (char 114)
Jun  2 14:53:55 WetPitsVQ systemd[1]: pijuice.service: Control process exited, code=exited status=1
Jun  2 14:53:55 WetPitsVQ systemd[1]: pijuice.service: Unit entered failed state.
Jun  2 14:53:55 WetPitsVQ systemd[1]: pijuice.service: Failed with result 'exit-code'.

Jun  2 14:53:55 WetPitsVQ systemd[1]: pijuice.service: Service hold-off time over, scheduling restart.
Jun  2 14:53:55 WetPitsVQ systemd[1]: Stopped PiJuice status service.
Jun  2 14:53:55 WetPitsVQ systemd[1]: Started PiJuice status service.
Jun  2 14:53:56 WetPitsVQ signalk-server[369]: wetpitsvq-for-signalk failed to start: EREMOTEIO: remote I/O error, write
Jun  2 14:53:56 WetPitsVQ signalk-server[369]: { Error: EREMOTEIO: remote I/O error, write
Jun  2 14:53:56 WetPitsVQ signalk-server[369]:     at Object.fs.writeSync (fs.js:735:20)
Jun  2 14:53:56 WetPitsVQ signalk-server[369]:     at Bus.i2cWriteSync (/home/pi/wetpitsvq-sk-plugin/node_modules/i2c-bus/i2c-bus.js:556:15)
Jun  2 14:53:56 WetPitsVQ signalk-server[369]:     at Oled._transfer (/home/pi/wetpitsvq-sk-plugin/node_modules/oled-i2c-bus/oled.js:136:25)
Jun  2 14:53:56 WetPitsVQ signalk-server[369]:     at Oled._initialise (/home/pi/wetpitsvq-sk-plugin/node_modules/oled-i2c-bus/oled.js:110:10)
Jun  2 14:53:56 WetPitsVQ signalk-server[369]:     at new Oled (/home/pi/wetpitsvq-sk-plugin/node_modules/oled-i2c-bus/oled.js:81:8)
Jun  2 14:53:56 WetPitsVQ signalk-server[369]:     at Object.<anonymous> (/home/pi/wetpitsvq-sk-plugin/display.js:18:15)
Jun  2 14:53:56 WetPitsVQ signalk-server[369]:     at Module._compile (module.js:653:30)
Jun  2 14:53:56 WetPitsVQ signalk-server[369]:     at Object.Module._extensions..js (module.js:664:10)
Jun  2 14:53:56 WetPitsVQ signalk-server[369]:     at Module.load (module.js:566:32)
Jun  2 14:53:56 WetPitsVQ signalk-server[369]:     at tryModuleLoad (module.js:506:12) errno: -121, code: 'EREMOTEIO', syscall: 'write' }
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]: Traceback (most recent call last):
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]:   File "/usr/bin/pijuice_sys.py", line 280, in <module>
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]:     main()
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]:   File "/usr/bin/pijuice_sys.py", line 197, in main
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]:     reload_settings()
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]:   File "/usr/bin/pijuice_sys.py", line 166, in reload_settings
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]:     config_dict = json.load(outputConfig)
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]:   File "/usr/lib/python2.7/json/__init__.py", line 291, in load
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]:     **kw)
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]:   File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]:     return _default_decoder.decode(s)
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]:   File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]:     obj, end = self.raw_decode(s, idx=_w(s, 0).end())
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]:   File "/usr/lib/python2.7/json/decoder.py", line 380, in raw_decode
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]:     obj, end = self.scan_once(s, idx)
Jun  2 14:54:00 WetPitsVQ pijuice_sys.py[644]: ValueError: Expecting , delimiter: line 8 column 2 (char 114)
Jun  2 14:54:00 WetPitsVQ systemd[1]: pijuice.service: Main process exited, code=exited, status=1/FAILURE
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]: Traceback (most recent call last):
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]:   File "/usr/bin/pijuice_sys.py", line 280, in <module>
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]:     main()
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]:   File "/usr/bin/pijuice_sys.py", line 197, in main
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]:     reload_settings()
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]:   File "/usr/bin/pijuice_sys.py", line 166, in reload_settings
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]:     config_dict = json.load(outputConfig)
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]:   File "/usr/lib/python2.7/json/__init__.py", line 291, in load
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]:     **kw)
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]:   File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]:     return _default_decoder.decode(s)
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]:   File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]:     obj, end = self.raw_decode(s, idx=_w(s, 0).end())
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]:   File "/usr/lib/python2.7/json/decoder.py", line 380, in raw_decode
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]:     obj, end = self.scan_once(s, idx)
Jun  2 14:54:01 WetPitsVQ pijuice_sys.py[706]: ValueError: Expecting , delimiter: line 8 column 2 (char 114)
Jun  2 14:54:01 WetPitsVQ systemd[1]: pijuice.service: Control process exited, code=exited status=1
Jun  2 14:54:01 WetPitsVQ systemd[1]: pijuice.service: Unit entered failed state.
Jun  2 14:54:01 WetPitsVQ systemd[1]: pijuice.service: Failed with result 'exit-code'.
Jun  2 14:54:01 WetPitsVQ systemd[1]: pijuice.service: Service hold-off time over, scheduling restart.
Jun  2 14:54:01 WetPitsVQ systemd[1]: Stopped PiJuice status service.
Jun  2 14:54:01 WetPitsVQ systemd[1]: Started PiJuice status service.`

Not sure when they started to appear. Any indications on what can be wrong?

tvoverbeek, 2019-06-02:
The pijuice service restarting happens when it tries to read the settings file '/var/lib/pijuice/pijuice_config.JSON' line 8. column 2 (ValueError: Expecting , delimiter).
Can you post the contents of your /var/lib/pijuice/pijuice_config.JSON ?
You probably need to do this as root, since /var/lib/pijuice/pijuice_config.JSON is owned by user pijuice and the directory is only accessible by users root and pijuice.

fencer, 2019-06-02:
JSON in the file seems to be invalid, there should be a coma on line 7. I wonder how this happened!
`{
	"system_events": {
		"low_charge": {
			"function": "SYS_FUNC_SYS_OFF_HALT",
			"enabled": true
		}
	}
	"system_task": {
		"enabled": true,
		"min_charge": {
			"threshold": "5",
			"enabled": true
		},
	}
}`

tvoverbeek, 2019-07-26:
Closing issue. If you do not agree, please reopen it.

closed 2019-07-26 by tvoverbeek