PiSupply/PiJuice #321: BP7X temperature

fencer, 2019-06-12:
Since recently the original BP7X battery is keeping a constant temperature of 40 degrees Celsius when kept in a ventilated area. This happens on firmware V1.2 and 1.3
Prior to this I could easily enclose the PiJuice and had no heating issues. What could be causing the battery to heat up?

shawaj, 2019-06-12:
Have you checked that the battery is making good contact to the battery
connector and is correctly seated?

fencer, 2019-06-12:
Yes it is correctly in place, to be honest I removed and inserted the battery a couple of times to be sure

tvoverbeek, 2019-06-12:
What is your 'temperature sense' setting?
On my PiJuice (not enclosed) with an environment temperature of 23ªC it shows
25ªC when on 'NOT_USED' and 38ªC for the other settings ('NTC', 'ON_BOARD' and 'AUTO_DETECT'.

Now after 15 minutes uptime (power connected to PiJuice USB, Battery at 98%, 1.3 Firmware) my temperature reading increased to 40ªC.
So 40ªC is not exceptional.

fencer, 2019-06-12:
I'm set to auto-detect. PiJuice has been running for a couple of hours, within a semi-enclosed environment and temp is reading 60deg Celsius. What is the safe operating temp of the battery? I'm targeting to completely enclose the device in a environment of aprox 40deg Celsius

tvoverbeek, 2019-06-12:
60 is about the limit for charging/discharging.
If you look at the BP7X temp limits in the profile:
Discharge between 0 and 60, Fast charge between 10 and 45.
Charging can occur above 45 but at reduced current.
See also this article: https://batteryuniversity.com/learn/article/charging_at_high_and_low_temperatures
Can you place the battery/PiJuice outside the 40 deg enclosure?

fencer, 2019-06-12:
No not really, because if the enviroment it will be in.  However as I said, it's been enclosed before and had no issue so there must be something that's causing the temperature to rise. Why should the battery heat up if not being charged / discharged? When the power is connected shouldn't it use the power supply instead?

fencer, 2019-06-17:
I'm still battling against the heat issue .... is there any way to make the PI juice only charge the battery when below a certain percentage and power the PI through external source when available so the battery does not heat up?

tvoverbeek, 2019-06-17:
Best is to power the PiJuice via the 5V GPIO, i.e. power the Pi from a 5V source.
This minimizes the heat dissipation by the power regulators on the PiJuice board.
You can disable the charging by command: `pijuice.config.SetChargingConfig(False, True)`
The second argument specifies if the setting should be stored in the PiJuice non volatile memory.
You can connect a user function to the 'low charge' event (has a settable threshold) to enable the charging again: `pijuice.config.SetChargingConfig(True, True)`
You will have to monitor the charge level yourself to disable the charging again when the charge level reaches a certain percentage.
Hope this helps a little bit to cope with the heat, especially now during the summer.

fencer, 2019-06-17:
Thanks for the tip! Sounds like a good idea. I can take care of the battery via my software np. 
By 5V GPIO do you mean J4?

tvoverbeek, 2019-06-17:
No, 5V to the RPi microUSB power input.
PiJuice is then powered via the 40 pin GPIO connector.

fencer, 2019-07-16:
Then I will loose the SW1 functionality right?

tvoverbeek, 2019-07-16:
@fencer Yes. Shutdown by longpress 1 will still work, but obviously the 5v stays on.
Therefore hard power on by sw1 will not work.
Only alternative then is to power via the PiJuice USB, but then you get the extra power dissipation by the voltage regulators.

closed 2019-09-16 by ChristopherRush

fencer, 2019-09-30:
Update: I have found out that most of the heat is originating from the PI and rising to the battery giving the impression the battery is heating up. I have now placed PiJuice on the side of the PI and the temperature is holding at an avg of 50°C