PiSupply/PiJuice #322: 5v LED Switch

fencer, 2019-06-13:
I am using J5 connected to an external switch/LED membrane: https://www.modmypi.com/raspberry-pi/sensors-1061/touch-sensors-1065/membrane-keybad-1-button-led

The -VE on the panel is  is common for the switch and LED. The aim is to have the witch turn on the PI and LED turn on with the PI. My concern is the common ground. Can I connect the +VE to the PI's 5V and the -VE and and return of the switch to J5 without doing any harm?

tvoverbeek, 2019-06-13:
I do not believe this will work

Here an extract from the PiJuice schematic showing how J5 and SW1 are hooked up.
Do not get confused by the SW2 on the schematic. This is the input on the MCU.
[image]
When SW1 is pressed the 3.3V is connected to the SW2 input of the MCU.
When you connect -VE to pin 2 of J5 you see there is a 10K resistor to earth.
With +VE connected to Pi +5V the LED will not draw enough current to go on due to the 10K resistor.

fencer, 2019-06-13:
thank you @tvoverbeek so what are the risks if I take 5v from the PI and GND from J5? attached a little sketch 
[image]

tvoverbeek, 2019-06-13:
There is no ground on J5.
If you connect it as in your picture you will have:
+5V - LED - 10K - GND
Hence only max 0.5mA through the LED. Probably not enough to light it.

fencer, 2019-06-17:
This appeared to initially work but after a few seconds the RPI would completely shutdown so as indicated by @tvoverbeek this cannot be done.

closed 2019-06-17 by fencer

tvoverbeek, 2019-06-17:
Not too surprising. When the LED is on the PiJuice MCU sees a continuous press of SW1 -> shutdown