PiSupply/PiJuice #327: Raspberry Pi 4B support

flobernd, 2019-06-24:
Hey there,

I already asked a similar question on the shop site, but the answer left some questions open. 

My original question was:
Does the PiJuice works with the new Pi4B (especially because of the new 3 ampere requirement). 

The answer was „yes“ because the PiJuice can output 3A in „pulses“ (what does that mean exactly?)

Is the PiJuice able to feed 3A to the Pi if I connect a 3A power supply to it (and not directly to the Pi)?

Best regards

shawaj, 2019-06-24:
The PiJuice has a 2.5A regulator that has a max output of 3A. So the answer you were given on the website is correct. However, the output completely depends on the battery you use, as charging or discharging LiPo batteries too quickly can cause overheating and safety issues. Hardware is often more nuanced than you are making out.

What is the application? I doubt you would need the full 3A in many circumstances anyway.

It will power the 4B no problem and we have tested it to work :-)

tvoverbeek, 2019-06-24:
Note the 3A are only needed when the CPU is on heavily loaded and the connected USB peripherals use 1.2A (!).
So without USB peripherals you are down to max 1.8A

flobernd, 2019-06-25:
Thanks for the answer!

The CPU load will be average to high most of the time with only a single USB storage device connected, so I should be fine for this project.

Have to test it I guess - as soon as there is new stock available. 

Best regards

closed 2019-06-25 by flobernd

ik0adr, 2019-07-10:
Hi all.
Tryed the pijuice hat on a new 4B+, but all I get is a I2C connection error.
I saw a comment about this on RPi forum, but no response.
Anybody here able to run PiJuice on a 4B+ under Buster?
Thanks..

shawaj, 2019-07-10:
add dtoverlay=i2c-bcm2708' in /boot/config.txt and it should work.

Do you have the link to the forum post?

ik0adr, 2019-07-10:
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=244535&p=1491479&hilit=pijuice#p1491479

A: PiSupply/PiJuice
Cc: ik0adr; Comment

tvoverbeek, 2019-07-10:
Had the same problem. Turned out when using the older i2c-bcm2708 module instead of the default i2c-bcm2835 everything works.
Put the following line on /boot/config.txt:
`dtoverlay=i2c-bcm2708`
and reboot. Works for me on a 4B.

shawaj, 2019-07-10:
@tvoverbeek I copied your email to me over to https://github.com/raspberrypi/linux/issues/3064 so hopefully we may get a response there.

flobernd, 2019-07-11:
> Put the following line on /boot/config.txt:
> `dtoverlay=i2c-bcm2708`
> and reboot.

Are there any drawbacks to this? I guess the i2c-bcm2835 version has some advantages over the i2c-bcm2708 otherwise it wouldn't be the default.