PiSupply/PiJuice #328: Dead battery?

tonecrafter, 2019-06-25:
I left my raspberry pi (3 B+) and PiJuice completely shutdown for several weeks. Now when I try and reapply power to the PiJuice it will not power on. If I power the Raspberry Pi via it's own micro USB connector the Pi powers on but nothing on the PiJuice. When I try and access the PiJuice using pijuice_cli.py the Pi cannot detect the PiJuice.

Is it possible I have killed my battery?

tonecrafter, 2019-10-24:
Hey Chris not sure why this has been closed as there was no resolution. Any thoughts on this one?

ChristopherRush, 2019-10-24:
Closed as there was no activity since June. Regardless of whether the battery is installed or not you should still be able to detect the PoJuice Hat with `i2cdetect -y 1` you should see the PiJuice at address 0x14. Did you use with any other HATs, peripherals etc Are there any LEDs lit up on the PiJuice? Try to reflash the Firmware.

tonecrafter, 2019-10-24:
I'll try this out and get back to you

RumoOr, 2020-02-08:
Same Problem for me. Battery isnt charging after Micro USB connection. SW1 push leads to no reaction. After battery take out and put in again, it might work for the next minutes/restart. But latest after next shutdown i got the same behavior.

tvoverbeek, 2020-02-08:
See also #413