PiSupply/PiJuice #329: Cannot install pijuice-gui under Raspbian Buster

rgw16, 2019-06-25:
I have burned a new SD card with the new raspbian issued with the Model 4.  When I try to install pijuice-gui as per the README documentation it fails to locate the package.

In the package list under stretch and jessie it falls between pigpio and pimoroni.  In Buster the list goes piclone, pimixer, pimoroni.

Will Pijuice be added to the new repository package list?

shawaj, 2019-06-25:
That is a question for the Pi foundation...will ping them now and see what they say

tvoverbeek, 2019-06-25:
I checked today (June 25) and the 1.4 packages are listed in both the Stretch and Jessie Packages lists, but not on Buster.
Did you run apt-get update for stretch or jessie?
For now use the Stretch version. Should work under Buster, but I have not tested this yet.

rgw16, 2019-06-26:
The stretch repository is not in buster by default. Should I add it to the
apt-get list?

Thanks
Rod

tvoverbeek, 2019-06-26:
Yes, you can add the line `deb http://archive.raspberrypi.org/debian/ stretch main ui` to
/etc/apt/sources.d/raspi.list.
The after an `sudo apt-get uodate` you should be able to install pijuice-gui and dependencies.
I recommend to remove the extra line in /etc/apt/sources.d/raspi.list after the installation to avoid
possible conflicts when updating software in the future.
Note: this will install the 1.4 version of the packages, since 1.5 is not available yet in the raspbian archive.

Alternatively you can install the 1.5 packages manually (sudo dpkg -i <package>.deb)
You can find them in https://github.com/PiSupply/PiJuice/tree/master/Software/Install.
Install pijuice-base first then pijuice-gui.
If it complains about dependencies you will have to install them first.
For pijuice-base:
Depends: python3 (>= 3.5), i2c-tools, python3-smbus, python3-urwid
For pijuice-gui:
Depends: python3 (>= 3.5), pijuice-base (>= 1.4), python3-tk

tvoverbeek, 2019-06-27:
See the main README for how to install on Buster for the time being.
Just tested on a fresh install of Buster.

closed 2019-06-27 by tvoverbeek

rgw16, 2019-06-27:
> Alternatively you can install the 1.5 packages manually (sudo dpkg -i .deb)

I tried the addition of the stretch repository and it allowed me to install
the pijuice-gui package.  Unfortunately it did not install on the path, or
anywhere I could find it.  After a reboot it did not start the demon.  I
removed the package.

I successfully downloaded the deb packages for 1.5 and installed them with
dpkg following your instructions above and the gui toolbar icon appeared
after a reboot.  All looks OK now.

Next step is to get my python program to log the voltage from disconnection
of the usb input until the shutdown from low voltage.  Then I'd like to do
the same exercise with the solar panel supplying input.

Thanks for the help.  I look forward to when the package becomes available
in the buster repository.

Regards
Rod

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=oa-4885-b>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=oa-4885-b>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>