PiSupply/PiJuice #330: Guide for setting up PiSupply batteries should be updated with the new profiles (1.3+)

Majkl578, 2019-07-01:
With the new firmware v1.3 new battery profiles arrived.
Some of them are "PJLIPO" profiles. I assume these refer to the _official_ PiSupply batteries? The profiles don't seem to be described more in detail anywhere (or did I miss something?).

The article about setting up batteries should be updated to reflect these new profiles since setting custom profiles manually could be quite dangerous (error prone with possibly severe consequences): https://learn.pi-supply.com/make/how-to-setup-connect-your-pijuice-battery/

ChristopherRush, 2019-07-09:
Will update docs and close, thanks.

Majkl578, 2019-07-12:
I also noticed that the actual profile data differ from the settings specified on the abovementioned page. For example PiJUice 12000mAh battery:
```
>>> config.GetBatteryProfileStatus()
{'data': {'validity': 'VALID', 'source': 'HOST', 'origin': 'PREDEFINED', 'profile': 'PJLIPO_12000'}, 'error': 'NO_ERROR'}
>>> config.GetBatteryProfile()
{'data': {'capacity': 12000, 'chargeCurrent': 2500, 'terminationCurrent': 350, 'regulationVoltage': 4180, 'cutoffVoltage': 3000, 'tempCold': 0, 'tempCool': 2, 'tempWarm': 49, 'tempHot': 65, 'ntcB': 3450, 'ntcResistance': 10000}, 'error': 'NO_ERROR'}
```

Is this by design?

closed 2019-09-16 by ChristopherRush