PiSupply/PiJuice #331: wake-from-halt

henr2011, 2019-07-07:
My Pi 3 running raspbian stretch is now restarting from poweroff every time I reboot, whether it is from choosing shutdown in GUI or sudo poweroff at command line. I have searched the forums and discovered it is an old issue but for some reason mine never did it before. My last significant change was updating pijuice to v1.5. Is there anything in the new  software/firmware which would affect this? It is annoying as I have to press and hold button 1 in order to turn it off now.

tvoverbeek, 2019-07-07:
I am assuming your SW1 switch off is the 'short long press' which executes a proper shutdown (Flashing LED 2 just before shutdown) and not the 'long long press' which removes power without properly shutting down. Correct?

Power input via PiJuce USB?
Firmware already upgraded to 1.3 from 28 Feb 2019?
Which Pi model and which Raspbian (Stretch, Buster, ...)?
What are your settings on the System Task tab?

The 1.3 firmware has some additional wakeup functions:
- When power is supplied via the GPIO from the Pi (power connected to Pi USB) then the pogo pin is not needed any more
- Additional wake up via IO2 on the J3 connector, but needs to be enabled from the gui/cli

Right now I cannot reproduce your issue.

henr2011, 2019-07-13:
Well this problem has disappeared so you can close this report. What I forgot to tell you was that in checking the pijuice hat config I discovered it was set for a custom battery, which I have never used. So I thought this could be evidence of a glitch during firmware update, anyway I changed it back to BP7x but also noticing that my 2nd and 3rd buttons had no effect I reprogrammed everything. It is now back to normal. I haven't had any problem like this with updating firmware before. Next time I will check all config afterwards.

Anyway, fixed now, thanks for your attention.

closed 2019-07-13 by henr2011