PiSupply/PiJuice #334: COMMUNICATION_ERROR for newly released PiJuice

dhvie, 2019-07-13:
Hi

I just received two new PiJuice hats and am trying to get pijuice-gui / cli to work. However I keep getting "COMMUNICATION_ERROR" in the GUI and when debugging further find that i2c throws [Err 121] Remote IO Error. This gets thrown in _Read in "PiJuiceInterface"

I have looked through the list of issues and have found a number of them dealing with this however couldn't find an issue with a resolution that worked for me. Here are my findings so far, hoping someone here knows that to do:

I have tried all of the below both on a Pi 3B and a Pi 4.
OS: Buster
Board: Pi 3B and 4
I2C: Verified enabled - i2cdetect output
root@pi4:~# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   

I have tried flashing to firmware 1.3 which is included with pijuice-gui. The flash is successful but does not solve the issue.

I have managed to get it to retrieve the firmware version via I2C but as for opening the cli/gui I always gets a comms error.
CLI error:
sudo pijuice_cli
[image]

I also noticed that the DIP switch 1 on the pijuice wasn't set to on, so I switched that the on side (1) to the up position as shown on the quick start guide.

I tried the above with both pijuice boards and both rpi boards. Does anyone have any idea what I could have missed here?

Thanks a lot for you help!

shawaj, 2019-07-13:
Take a look at https://github.com/PiSupply/PiJuice/issues/327

shawaj, 2019-07-13:
We are also working on this here https://github.com/raspberrypi/linux/issues/3064

dhvie, 2019-07-13:
Thanks for the reply @shawaj 
I tried adding the overlay as well as doing a rpi-update and retesting before each step however the error has persisted so far.

shawaj, 2019-07-13:
The PiJuice is behaving correctly if there is entries on 14 and 68 i2c addresses.

Could you try flashing the firmware manually using the process described here:

https://github.com/PiSupply/PiJuice/blob/master/Firmware/README.md

If that still doesn't work, I would recommend starting from a fresh version of buster, then add the config.txt line, install PiJuice and enable i2c etc and only then attach the PiJuice HAT itself.

Let us know how you get on.

shawaj, 2019-07-13:
Also @dhvie have you restarted after making the changes, the rpi-update and config.txt changes will only be live after a restart

dhvie, 2019-07-13:
Yeah I figured a fresh start would be good so I did the original steps from a fresh install of buster. Will try flashing firmware and report back.

shawaj, 2019-07-13:
Thanks very much

dhvie, 2019-07-13:
I have not managed to get it to work. However you can't run the gui or cli via the normal executable (even using sudo) as this will always say that the pijuice hasn't been detected.
If I start it with "sudo python3 /usr/bin/pijuice_gui.py" or similar for the cli, it work however.
I can't manage to get the icon into my system tray yet.

tvoverbeek, 2019-07-13:
@dhvie Are you running as user pi or as root?
The executables require that the user belongs to the group pijuice.
The install automatically adds the defualt user (uid 1000, normally pi) to the pijuice group.
If you have changed to an other user try to add the user to the group pijuice, then it should work (I hope)

dhvie, 2019-07-13:
I added my user to pijuice and rebooted but still no change unfortunately :(

tvoverbeek, 2019-07-13:
@dhvie You are obviously not running as user pi. Then you also need to add your user to the i2c group to get access to the i2c device. Try to run i2cdetect as your non-privileged user (not as root). If you brlong to the i2c group you do not need to run i2cdetect as root.

dhvie, 2019-07-13:
@tvoverbeek ah yes adding my user to i2c did the job. Thanks a lot!

closed 2019-07-13 by dhvie