PiSupply/PiJuice #336: PiJuice doesn't work well with RPi4B in spite of statements otherwise

DJenner8, 2019-07-20:
I installed a PiJuice on my RPi4B.  It wouldn't communicate until I modified config.txt to load the older overlay.  Then, it appeared to run OK.  I was powering system through PiJuice using older official adapter.

However, when I ran a CPU intensive app for a short time, the RPi crashed.  I was monitoring the system with Cronky and CPU was 97% for less than 2 seconds.  Crash.

I switched to powering directly to the RPi4B with the new official adapter.  When I shutdown, the RPi will shutdown, but after a few seconds, it restarts.  If I remove the power, it reboots, so using the PiJuice for battery backup is useless.  If I shutdown under battery power, it reboots.

Unless I am missing something, the PiJuice is not going to work for the RPi4B.

tvoverbeek, 2019-07-20:
If you are running with the default 1820 mAh BP7X battery then the PiJuicd will not be able to supply the needed current for a Pi4B under heavy load and USB peripherals attached.
Max current which can be supplied to the 5V GPIO by the BP7X battery is certainly less than 1.5A.
Hence the crash you saw. You will need to attach a bigger battery.

When powering the RPi4B directly the restart after shutdown will happen if you have enabled the wakeup-on-charge in the settings for system task.
Similar for reboot after removing power. Check the setting for 'No Power' in the system events tab.

Anyway, for a Pi4B with a high CPU load and/or power hungry peripherals (USB3 SSD e.g.) you will definitely need a larger capacity battery than the BP7X.

DJenner8, 2019-07-21:
Ton, Thank you for the information.  I moved to a bigger battery, which I already had on hand.  That seems to have solved my crashing problem on the 4B.  I am still working on my 3B+ system to make sure it is behaving correctly with a PiJuice/BP7X running the same SD Card with Buster.  So far, so good.

reissken, 2019-08-03:
I am getting a DATA_CORRUPTED showing in all the config fields on a pi 4. Any advice for that?

tvoverbeek, 2019-08-03:
@reissken Known problem on the Pi4 due to change in the I2C hardware. For now fall back on the older bcm-2708 module.
Include the following line in /boot/config.txt `dtoverlay=i2c-bcm2708` and reboot

daphatty, 2019-08-20:
Regarding the aforementioned DATA_CORRUPTED behavior, I saw something similar while using a Pimoroni FanShim, a GPIO extender, and the PiJuice on an RPi4. DATA_CORRUPTED would appear on screen for a few seconds, briefly switch to the normal data points, and back again. Figured I would mention it here since I won't be the only one using a fan in between the PiJuice and RPi4 CPU.

jscoobyced, 2019-10-29:
> @reissken Known problem on the Pi4 due to change in the I2C hardware. For now fall back on the older bcm-2708 module.
> Include the following line in /boot/config.txt `dtoverlay=i2c-bcm2708` and reboot

I saw that in another issue here and that indeed solved my DATA_CORRUPTED issue.
I also enabled I2C in raspi-config but not sure it helps.

I have the same issue as this post, my battery level goes down on a RPi4B. I'll order a bigger battery. Thanks all, I should be good soon!

tvoverbeek, 2019-10-29:
Closing the issue now. Reopen it if you do not agree, or raise a new issue if it is a different problem.

closed 2019-10-29 by tvoverbeek