PiSupply/PiJuice #337: PiJuice fails to charge batteries, no power draw over internal MicroUSB

jbernra, 2019-07-21:
I tried to get the UPS to work with a Pi 4 and 3+, both running the most recent Noobs installation and the PiJuice Software.
Although I have been trying now for two days I can't get my PiJuice to charge the batteries I got for it (BP7X and A855XL). Furthermore it doesn't seem to be able to draw power from the MicroUSB on the PCB. 
It recognises a plugged in cable but the power meter doesn't show a current being drawn, depleting the batteries further.
Is this a known issue or may it be possible that I received a faulty unit?

shawaj, 2019-07-21:
It sounds like you maybe don't have charging enabled from this menu shown here https://github.com/PiSupply/PiJuice/blob/master/Software/README.md#pijuice-hat-general-config-menu

jbernra, 2019-07-21:
Hi shawaj, thanks for your quick reply!
I have checked the menus at least three times charging it definitely enabled, but this wouldn't explain why it fails to draw power via the integrated Usb port. 
If I remove the battery and attach my 2 Amp Usb power brick to the inlet I can't get either one of my raspberries to start. Only when a battery is present it will start, provided the used battery is sufficiently charged and will run down this charge. It discovered this only when the first battery was nearly completely depleted.

shawaj, 2019-07-21:
Are you holding the power button on when you try to start it with no battery?

Could you send a screenshot of the main PiJuice menu and the HAT config main menu as well?

jbernra, 2019-07-21:
Yes I have done that, but it didn't make a difference.

Find the screenshots attached.
[image]
[image]

tvoverbeek, 2019-07-23:
@jbernra On your screenshots I see the status of GPIO power input as BAD.
BAD means the power input is too weak and therefore the charge control chip disables charging.
What power input do you use for the RPi? Is it a Pi4B?
The Pi4B is more power hungry than previous RPis and that leaves less power available for the PiJuice.

You might want to try to connect straight to the PiJuice micro-USB with the RPi off to get the charging started (This will of course not work with a Pi4 USB-C power supply, you need a power supply with a micro-USB connector)

jbernra, 2019-07-23:
@tvoverbeek Yeah I noticed that too and can't really explain that, because the power supply should be able to provide enough juice. For reference see the attached images.
I did all my last tests with the Pi3B+ so it should definitively work.

Connecting it directly doesn't work, because the PiJuice won't draw any power for some reason, see the attached pictures

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

tvoverbeek, 2019-07-23:
@jbernra Is this a new PiJuice?
Stupid question (maybe): Have you removed the plastic strip protecting the BP7X contacts?

jbernra, 2019-07-23:
@tvoverbeek Yeah it is, I got a couple of days ago and yes the strip has been removed ^^

tvoverbeek, 2019-07-23:
Do you have an other power supply? The recommended one for Pi3B+ is a 2.5A one.

jbernra, 2019-07-23:
The white one that you can see in the pictures along the power meter is a 2.4A charger, but given that the processor load on the pi was extremely low and apart from a keyboard receiver there have been no peripherals attached I don't see why that extra power should be needed. (As the Pi draws only around 40mA in this configuration)
I have left the PiJuice stacked on the Pi while it was shutdown to allow the battery to be charged without the Pi drawing any significant amount of power, but the meter didn't show any change in the amount of power drawn. (0.05mA)

tvoverbeek, 2019-07-23:
Last thing to try is to reset the configuration to default:
```
Dual long press of SW1 and SW2 for 20 seconds will reset PiJuice HAT configuration to default. This applies to the MCU configuration only.
```
Note SW1 and SW2 are the two switches at each end. The order on the pcb is SW1, SW3, SW2.
WHen the reset starts the LEDs are flashing for a while.

If this does not help, please contact Pi Supply (or your supplier) directly with reference to this issue. Maybe the unit is defective.

jbernra, 2019-07-23:
Ok I did that and it sadly didn't fix the issue, i'll contact the supplier.
Thank you very much for the time you invested into this, your help has been much appreciated! ☺️

shawaj, 2019-08-05:
@jbernra we received the returned unit and we are shipping a replacement to you tomorrow.

Hopefully will have more luck with this next one!!

shawaj, 2019-08-06:
Also FYI I sent the broken one to @mmilann for further investigation

ChristopherRush, 2019-10-23:
@mmilann Did you look at the cause of this issue?

closed 2019-10-23 by ChristopherRush

oalkhayat, 2020-02-26:
Has a root cause been discovered for this issue? I've observed similar issues.

decoder247, 2021-07-26:
I'm experiencing the same as @oalkhayat, particularly that the USB micro power input has a BAD status. Is there any way to remove the power input limit set for the charge control chip that @tvoverbeek mentioned in his comment, or is this not possible? - "_BAD means the power input is too weak and therefore the charge control chip disables charging._"

jkleiman, 2022-06-03:
Try pulling the battery out and putting it back in