PiSupply/PiJuice #338: SMBus and I2C commands / addresses for remote battery interrogation

alinnz, 2019-07-23:
HI can anyone help with the commands for the SMBus I want to display the values from the Pi Juice on a dashboard currently have it running but the values are wrong ?

tvoverbeek, 2019-07-23:
@alinnz Are you using the functions available in pijuice.py? Or are you using i2c commands directly?
Anyway the code in pijuice.py gives you the i2c commands to use.
However voltage, current, etc. values are raw and need to be scaled to engineering values. You can look in the source code for pijuice_gui.py or pijuice_cli.py to see how this is done.
If you give me some more details, I might be better able to help you.

AlanHares, 2019-07-23:
@tvoverbeek Thanks for the quick reply, I am using the I2C commands directly, I want to display the information on a Node-Red Dashboard I have done this with other battery management, but when I use the standard SMBus commands  the results don't match the figures actually displayed in the Pi Juice config panel, not sure how to look at pijuice.py ?
Thanks :)

tvoverbeek, 2019-07-24:
Can you give me some examples of your I2C commands and its results compared to the PiJuice config panel?
Some  info: Voltages and currents are returned as 2 bytes in the order low byte, high byte.
Battery current and IO current have to be interpreted as signed 16-bit.
Those 16-bit values have to be divided by 1000 to get voltage (V) and current (A).

AlanHares, 2019-07-24:
Thanks for the info I will try later to get some examples but I did find the Hex addresses in the pijuice.py code thanks.

AlanHares, 2019-07-25:
@tvoverbeek  Ok now I have the correct command addresses its all working great, I just have a question about the Battery Temperature it doesn't show on the Pi Juice Config GUI ? but I am getting 33-32 Degrees straight from the low byte and 0 on the high byte from address 0x47 is there any conversion needed ?

Thanks again :)

tvoverbeek, 2019-07-25:
If you do not see the battery temperature in the GUI you are running an older version of the software.
If you are on Stretch or Buster you can upgrade to the 1.5 version of the Raspbian packages which will show the battery temperature in the GUI/CLI. 33-32 degrees is normal, no conversion needed.

AlanHares, 2019-07-25:
Thanks for all the help its all running great now, if I wanted to get the "Status" i.e Normal Charging etc.... I the address 0x40 I get 53 returned is this correct and is there a key to show what the returned codes mean?

Sorry for all the questions but our system is running completely remotely so the more info we get remotely the better :)

Thanks again

tvoverbeek, 2019-07-25:
Look at the GetStatus function in pijuice.py how the status byte is broken down:
```
            status = {}
            status['isFault'] = bool(d & 0x01)
            status['isButton'] = bool(d & 0x02)
            batStatusEnum = ['NORMAL', 'CHARGING_FROM_IN',
                            'CHARGING_FROM_5V_IO', 'NOT_PRESENT']
            status['battery'] = batStatusEnum[(d >> 2) & 0x03]
            powerInStatusEnum = ['NOT_PRESENT', 'BAD', 'WEAK', 'PRESENT']
            status['powerInput'] = powerInStatusEnum[(d >> 4) & 0x03]
            status['powerInput5vIo'] = powerInStatusEnum[(d >> 6) & 0x03]
```
I assume your 53 is decimal (=0x35)
Decoding 0x35 then gives:
isFault: True
battery status: 'CHARGING_FROM_IN'
powerInput: 'PRESENT'
powerinput5vIo: 'NOT_PRESENT'

Similar for other variables. The pijuice.py source is here: https://github.com/PiSupply/PiJuice/blob/master/Software/Source/pijuice.py

AlanHares, 2019-07-25:
So 0x35 = true gives the 3 results? I looked at the code but cant see any reference to 0x35 ?
what would be the hex codes for other scenario's ?
Sorry not much of a programmer except for Node-Red 

Thanks

tvoverbeek, 2019-07-26:
0x35 corresponds to 00110101 in binary.
The 2 most significant bits (here 00) are the powerInput5vIo status.
This can be 00,01,10,or 11 (0,1,2,3 in decimal). Used as index into the powerInstatusEnum 00 coresponds to 'NOT_PRESENT'.
The next 2 bits (here 11) are the powerInput status. With 3 as index we get 'PRESENT'.
The following 2 bits (here 01) dre the battery status. With index 1 this gives 'CHARGING_FROM_IN'.
The following bit (here 0) is the button status (0=False, 1=True)
The last (least significant) bit (here 1) is the Fault status.
There are plenty of Javascript and Node-RED bit manipulation packages
Good luck with your dashboard

alinnz, 2019-07-26:
That’s an awesome explanation, thanks so much I’ll be searching out the nodes today 

Thanks again 😊

Alan 
Smart Controls Ltd
EV Charging Solutions
021 764 205

AlanHares, 2019-07-28:
@tvoverbeek all working great now thanks :) the I/O Current fluctuates between 1 and 2.5 amps is this what you would expect to see ?